atsamd-rs/atsamd #6: Publishing of arduino_mkrzero

djmcgill, 2018-09-03:
I'm a bit confused by the structure of the repo - the crates are entirely separate according to the rust/cargo model right?

Do you have any plans to publish the arduino_mkrzero crate as the owner of the repo or shall I as its primary author?

wez, 2018-09-04:
It's worth documenting this stuff, so here's a quick stab at it:

The crates are all in one repo to make it easier to atomically make changes across all of them and run the CI in a single PR.  Without that grouping, sweeping changes are more challenging to make, review and test.

When it comes to publishing the individual crates, each `Cargo.toml` file needs to be prepped and published as you would do with a normal crate-per-repo.

There is a small difference in that the board crates do something like this:

```
[dependencies.atsamd21-hal]
path = "../hal"
version = "~0.1"
default-features = false
```

This allows building in-tree and also specifying the version number for the crates build outside of the tree.  When the interfaces change, these versions need to get bumped as you might expect.

What I think needs to happen to publish `arduino_mkrzero` is that we most likely need to publish the updated peripheral crate(s) and the hal crate so that you can publish `arduino_mkrzero`.

Regarding crate ownership: I don't feel overly possessive and have granted you co-ownership of the existing crates so that you're unblocked from publishing.  It would be nice if you reciprocated for `arduino_mkrzero` when you create it so that we are mutually unblocked from prepping and pushing releases in the future.

I think what I did last time was just to work through from the peripheral crate and publish it, then moved to the hal crate, then to the board crate, then once all were published I think I pushed the commit with the version bumps to the repo.

djmcgill, 2018-09-11:
Sorry for the delay I'm looking at this again now. I don't believe that there have been changes to the peripheral crates right? Just HAL upwards?

wez, 2018-09-13:
There were some changes in the svd generated crates.  I went ahead and bumped some versions and published the SVD crates, the hal and the metro_m0 crate.  I believe that you're all set to just publish `arduino_mkrzero` when you're ready; you needn't worry about any of the other crates in this repo.

djmcgill, 2018-09-17:
Okay, published and ownership invite sent:
https://crates.io/crates/arduino_mkrzero

closed 2018-09-17 by djmcgill

zklapow, 2018-09-23:
I also published the `samd21_mini` crate FWIW, and added you as an owner @wez