atsamd-rs/atsamd #7: concat_idents in define_pins macro

djmcgill, 2018-09-03:
Good news and bad news: 
I found a crate (`mashup`) which lets you generate identifiers, so the macro call now looks like
```
define_pins2!(
    struct Pins,
    target_device: atsamd21g18a,

    /// Digital 0: PWM, TC
    pin d0 = a22,
```

But the macro now looks like https://github.com/djmcgill/atsamd21-rs/blob/mashup-iden-test/arduino_mkrzero/src/lib.rs

Is this a solution you'd be interested in or are we tying ourselves in knows just to save a small amount of duplication (which gets compile checked anyway)?

wez, 2018-09-04:
Oooh.  I'm a bit on the fence because the new dependency doesn't seem to add a huge amount of value and introduces some magic with the max recursion depth thingy... but it is nicer.
Does this make the compilation times dramatically longer?
Did it strike you as overly complicated to produce that macro vs. the vanilla rust macro experience?

wez, 2018-09-04:
let's use the new macro; I've a feeling that we're going to grow more board crates and it will help keep the boilerplate lower.

djmcgill, 2018-09-04:
Closed via https://github.com/wez/atsamd21-rs/pull/9

closed 2018-09-04 by djmcgill