atsamd-rs/atsamd #11: USB Support

wez, 2018-09-14:
This issue is tracking the implementation of USB.

 - [ ] USB Device crate atomics don't compile on thumbv6m https://github.com/mvirkkunen/usb-device/issues/2

zklapow, 2018-09-25:
FWIW I spent some time getting the USB stuff to work with RTFM (which could help with eliminating the globals) but found it hard to get both the inner references (i.e. device needs a ref to the serial object) and then return those same objects from init (because the objects constructed in `init` did not have `static` lifetimes). I am sure there is some magic here that would allow RTFM to work that I am just missing though I am not sure whether RTFM counts as solving the globals issue.

fralalonde, 2019-04-26:
Not sure what the official status is on this, I just want to report that I've tried porting the `usb_serial` example from metro_m0 to a trinket_m0. To do so, I've un-commented the `usb` feature from hal and enabled it from trinket_m0. I was expecting compilation issues with atomics (?) but the build went fine - using nightly. Loading the resulting binary resulted in my Linux host reporting `cdc_acm 5-2:1.0: failed to set dtr/rts` and the trinket immediately locking up. 

I've switched clock init to use the internal oscillator (trinket_m0 has no external osc?) and am now getting EILSEQ(-84) and EPROTO(-71) errors on dmesg log upon trinket reset. I'm assuming some form of miscommunication is now occuring. Also the board no longer locks up. Yay, progress?
```
[1490808.087147] usb 5-2: new full-speed USB device number 22 using uhci_hcd
[1490808.251255] usb 5-2: device descriptor read/all, error -84
[1490808.373836] usb 5-2: new full-speed USB device number 23 using uhci_hcd
[1490808.536213] usb 5-2: device descriptor read/all, error -71
[1490808.537160] usb usb5-port2: attempt power cycle
[1490808.973926] usb 5-2: new full-speed USB device number 24 using uhci_hcd
[1490809.011214] usb 5-2: device descriptor read/all, error -71
[1490809.340593] usb 5-2: new full-speed USB device number 25 using uhci_hcd
[1490809.371206] usb 5-2: device descriptor read/8, error -71
[1490809.504211] usb 5-2: device descriptor read/8, error -71
[1490809.610493] usb usb5-port2: unable to enumerate USB device
```

Anyway, not having a debugger (yet), using LED blinks as a heartbeat and being otherwise way out of my league I'm kind of stuck there for now but I thought I would just drop by and leave my cool bro story for others to read while I'm sadly going back to regular Arduino `ino` coding to achieve some ordinary success with my USB MIDI project. 

As a general note, I feel embedded Rust is shaping up nicely and I'm still quite excited about it. Good work!

twitchyliquid64, 2019-09-13:
I'm interested in working on this, but as a newbie to rust its best if I'm told what to do. How can I help?

sajattack, 2019-09-14:
This is one of those thing's that's left over from when @wez was running the project and frankly I'm scared to touch it lol.

wez, 2019-09-15:
I've been a bit too busy to pick this up in a while.  Things have matured a bit more in the upstream usb-device crate so what I think needs to be done in this one now is:

* Update to the latest usb-device code; there have been a few API changes that will require some changes
* Revise https://github.com/atsamd-rs/atsamd/blob/master/boards/metro_m0/examples/usb_serial.rs to use the now standalone CDC-ACM serial implementation crate https://github.com/mvirkkunen/usbd-serial

With that done, we'll have a relatively low level support for USB on this class of devices and can consider actually enabling that code when publishing to crates.io, but there is still more to be done to make practical USB device implementations:

* USB HID support (near and dear to my heart for keyboards)
* USB storage device support to expose eg: on-board SPI based flash storage, or perhaps UF2 bootloader

I see these as requiring some work in the upstream usb device crate with appropriate atsamd-specific support happening in this crate.

There's also a distinction between low-level support (eg: it basically works) and ergonomic high-level support (eg: there's a nice way to respond to USB HID events and embed support in a consuming application crate).

Beyond the bus implementation in this crate, most of the work to be done is generic USB stuff that should happen in the usb-device project.

twitchyliquid64, 2019-09-16:
I'll take a look at updating the samd21 crate, but I don't have hardware so it will need someone else to test.

I'll also try and get it working for samd51, which I do have hardware for.

twitchyliquid64, 2019-09-17:
Looking at the samd21 code, it looks very similar to what is required for samd51. I expect there's a good amount we can put in the common crate.

twitchyliquid64, 2019-09-22:
This is slowly coming along [here](https://github.com/twitchyliquid64/atsamd/tree/master/hal/src/samd51/usb). Some notes/changes for comment:

 - Instead of only exposing an implementation of the `usb_device::bus::UsbBus` device, I'm going to expose a lower-level (no state, no memory management) API `UsbController`, which the `UsbBus` implementation will be built off. 
 - I'm refactoring the `ep!` macro to collect all the registers for a given endpoint into a single structure, `EndpointRegs`, which will provide a nicer internal API over mutating the state of the endpoint registers.
 - Aside from HAL layers, move away from using the term _Bank_ to describe one endpoint in an endpoint pair, as I couldnt find any reference to it outside of datasheets.
 - I'm using a single lifetime across all objects from `UsbController` downwards, and instantiating them all at the same time. This allows us to avoid `Ref*` and `Mutex` boxing.

twitchyliquid64, 2019-09-27:
Heya @wez, how did you get the `dbgprint!` macro to dance your tune?

I made the mistake of writing the samd51 driver all at once, so it attaches but doesnt enumerate. I'm thinking it would be nice to have a printf out a SERCOM UART, but if thats something youve already done I'll gladly copypasta :)

BenBergman, 2019-11-29:
@twitchyliquid64 Did you make any progress on the samd21 port? I've got some devices I'd be willing to test with.

twitchyliquid64, 2019-12-02:
@BenBergman no - haven't ported samd21 USB support over to the 0.3 version of usb-device. The original code (which ran from an earlier version) should still be usable.

In terms of working on the USB peripheral, I've been side-tracked by other projects as of late, but hope to get back to this + fixing the other bug soon.

BenBergman, 2019-12-02:
I tried compiling the Metro-M0 usb-serial example but it is failing to compile for me, so I'm guessing some other change broke something. I poked at it a bit but low level USB is a bit beyond me and the limited time I have to focus on it.

Let me know when you get back to it and I can test on a few Adafruit M0 boards I have lying around.

closed 2020-01-25 by sajattack