atsamd-rs/atsamd #13: Story around hal re-exports

djmcgill, 2018-09-19:
In `arduino_mkrzero`, I just copied the same structure of `pub use` etc that existed in the other board libraries.

However now I'm actually trying to use them in an application I'm running into difficulties. The`atsamd_hal` and `embedded_hal` modules aren't actually exported, only their contents which behaves oddly. You can see in [this search](https://wez.github.io/atsamd21-rs/atsamd21g18a/arduino_mkrzero/hal/index.html?search=OutputPin) that the only OutputPin type exported by the board library is `arduino_mkrzero::prelude::_embedded_hal_digital_OutputPin`.

The problem goes away if the user includes `embedded_hal` directly in their own `Cargo.toml` but if that's the expected case then the re-exports are still useless.

Any suggestion? The simplest solution would be to drop all the re-exports and the user is expected to import the `embedded_hal` crate themselves in order to use the board crate. Or we could `pub use atsamd21_hal as hal` (i.e. NOT a glob import) and then users could access `arduino_mkrzero::hal::digital::OutputPin`.

In short the current solution doesn't really suit our (users') needs and we should think about a better one :)

Either that or I'm using the board library wrong in which case I suggest that the documentation be improved to make things clearer.

wez, 2019-01-05:
With my recent messing around with the 2018 edition, I think these glob imports confuse rust a bit and make it harder to reason about what is in which namespace, so I'm in favor of tidying things up.

I propose eliminating all glob exports, except in prelude modules that are re-exporting eg: `embedded_hal::prelude::*`.

I think the board crates should re-export `atsamd_hal as `hal`.
For clarity, I think `atstamd_hal` should `pub extern crate embedded_hal;` rather than `pub extern crate embedded_hal as hal;` (as it does currently), so that consumers can refer to the traits as `hal::embedded_hal` rather than the harder to intuit `hal::hal` that we currently have.

I also propose that the configured hal export the pac crate as `target_device` to help minimize conditional feature checks downstream.  I think we could *also* export it with its native crate name for potentially more advanced uses, but I don't think it is strictly required.

djmcgill, 2019-01-05:
Going back to first principles, what is the advantage to re-exporting at all? In all of my code I ended up importing the embedded_hal crate via cargo anyway, would this be able to eliminate that?

wez, 2019-01-07:
For me, re-exports mean that I have fewer details to worry about; in my application crates I just need to add a dep on the board crate and then I don't have to worry about copying and updating the versions of the crates that it depends upon; I can just reference those via the re-exported names.

That doesn't mean that my code can ignore API changes, but it does reduce some mechanical/boilerplate changes and helps to (partially) insulate the application from rather inscrutable errors from time to time as the deps in the embedded ecosystem change.

closed 2019-12-29 by sajattack