atsamd-rs/atsamd #20: RFC: How should Adafruit's Feather M0 boards be organized

BenBergman, 2018-10-18:
Adafruit's Feather line has something like 5-10 M0 variations depending on how you want to count the different radio variations. Should there just be one board module for all Feather M0 boards or should there be a new board folder for every single one?

BenBergman, 2018-10-18:
I should add that the boards all have identical pin aliases for the exposed pin. They only differ in whatever the added external peripheral is (wireless radio, SD card, etc).

sajattack, 2018-10-18:
I think just one for the Feather M0 Express and then people can use that with all the feather variants.

sajattack, 2018-10-18:
I have a Feather M0 express coming in the mail tomorrow :)

BenBergman, 2018-10-19:
Yup, PR #22. Note that the Feather M0 Express is the only one with an SPI flash chip, so maybe the Basic Proto one is the one that should act as the base?

wez, 2018-10-19:
Targeting the basic proto as a base seems like a sane plan and should avoid proliferation of board crates.

We can perhaps add little helper crates for the peripherals; for example, the blefriend module is available as a breakout as well as bundled up with an M0 Feather.  A crate for managing the SDEP protocol to speak with that seems like a fine idea and doesn't strictly require that we define a separate feather board crate just for that use case.

I know that @sajattack is interested in UF2 support, so it is probable that we'll end up with board crate for the Express variant to tie together the right features and easier to use out of the box.

Let's keep a watch on the amount of boilerplate in these crates and be prepared to use some macros or other techniques to keep it small.

BenBergman, 2018-10-20:
That sounds great to me.

I'm also wondering if the SPI/I2C/etc initialization functions can somehow get a more ergonomic interface, perhaps via macros. From my I2C example in the `itsybitsy_m0` module:

```rust
let i2c = hal::i2c_master(
        &mut clocks,
        KiloHertz(400),
        peripherals.SERCOM3,
        &mut peripherals.PM,
        pins.sda,
        pins.scl,
        &mut pins.port,
);
```

Setting that up for the first time, I had a hard time trying to figure out where to get `SERCOM3` and `PM` from. Thankfully clocks was pretty straight forward, but it seems like it is always the same part of `peripherals`. Similarly the port will always be the same and, since the initializer is for a specific I2C configuration, the pins will always be the same too. This just leaves the speed as the only thing the user would want to set specifically.

Maybe that is just a necessary complexity for now given the current state of the `embedded-hal` and the fact that we can't borrow `peripherals` once `clocks` has been borrowed out of it, but if there is something we do to make it a bit more ergonomic that would be excellent.

sajattack, 2018-10-23:
@BenBergman I have now tested your feather_m0 PR as well. Great work!

closed 2019-12-29 by sajattack