atsamd-rs/atsamd #21: Default to OpenOCD

sajattack, 2018-10-18:
OpenOCD seems to be the standard these days for rust-embedded. See the [rust-embedded](https://rust-embedded.github.io/book/start/hardware.html) book for example. Until we get UF2 working I think it makes more sense to use OpenOCD rather than JLink.

BenBergman, 2018-10-19:
Have you been using OpenOCD or JLink with your boards? What extra hardware is required for that? I've been using the `bossac` loader that comes with Arduino via the USB bootloader Adafruit ships on their M0 boards.

sajattack, 2018-10-19:
I haven't tried yet. It's documented in the README though and the .gdbinit defaults to JLINK. I think you could use ST-LINK as well but not sure. Real JLinks are expensive but of course there are chinese clones. 
https://shop.segger.com/DebugProbe_s/40.htm?searching=Y&sort=1&cat=40&show=30&page=1
https://www.aliexpress.com/item/JINSHENGDA-High-Speed-J-Link-JLink-V8-USB-ARM-JTAG-Emulator-Debugger-J-Link-V8-Emulator/32846502604.html

wez, 2018-10-19:
FWIW, I've never had a successful debug session with OpenOCD, whereas JLink always works.
OpenOCD worked to flash the board, but when I need to debug, it's JLink or don't bother.

The JLink EDU Mini is a lot cheaper than the full JLink and works just as well as its bigger brother (although the data transfer rate is deliberately throttled: you'll notice it when flashing):
https://www.adafruit.com/product/3571

wez, 2018-10-19:
I'm neutral on changing the default to OpenOCD, but would like it to remain convenient to use JLink, so keeping a jlink flavored .gdbinit file in the repo would be strongly desirable for me.

Is it possible to tell which is in use and just have one .gdbinit file that works with both?

The bobbin CLI does start the JLink gdb server on the same TCP port as the OpenOCD server, so that's one step in the right direction.

BenBergman, 2018-10-19:
Some quick googling seems to indicate that the ST-LINK doesn't normally support ATSAMD chips, but the [Black Magic Probe](https://github.com/blacksphere/blackmagic) firmware can convert an ST-LINK (or even a Blue Pill dev board) into something that can handle both STM and ATSAMD21 chips (though it seems ATSAMD51 isn't supported yet). I've got a few Blue Pills in the mail, so I'll have to try that out.

wez, 2018-10-20:
FWIW, I think there are a couple of different use cases that we should consider here:

1. Building and flashing
2. printf debugging via eg: "serial" aka USB CDC
3. "real" debugging with debug probes

Item 1) should be easy for casual or new-to-embedded users to do, as most boards have some convenient boot loader and tooling for this.
Item 2) is currently blocked on USB support, but once we have that it should also be easy to use.
Item 3) is really for power users that can tolerate some manual local configuration.

The overall situation is a bit more nuanced when we consider what people are building and how:

* Examples in this repo are built out of this repo
* End-user applications will be built in crates/repos of their own

There are a few permutations to consider and/or document.

We can and should document using OpenOCD in the main readme; it's largely absent because I haven't had a terribly successful time with it and didn't feel like endorsing it.

I also think we should consider documenting using `bobbin` tool; you'll see some configuration for that in the `metro_m0` crate.  I think it does a nice job of managing the complicated and fiddly bits, and it integrates nicely with building.

Check out these docs: https://github.com/bobbin-rs/bobbin-cli/#installation

closed 2019-12-29 by sajattack