atsamd-rs/atsamd #27: Implement RNG abstraction

BenBergman, 2018-11-24:
The embedded-hal crate provides a trait for an RNG but I don't believe it is implemented yet. I haven't had a chance to research it yet but wanted to get the feature tracked.

https://docs.rs/embedded-hal/0.2.2/embedded_hal/blocking/rng/trait.Read.html

sajattack, 2019-01-04:
I think only the '51 has the rng, not the '21

BenBergman, 2019-01-05:
Oh, interesting. I guess I need to read the LSB of a floating ADC a few times to get a random number or something... Or is there something generic that we could provide with the hal as a psuedo RNG?

sajattack, 2019-01-05:
I think circuitpython just uses a software PRNG.

sajattack, 2019-03-03:
Looks like the best way to do this is implement the RngCore trait. https://docs.rs/rand/0.5.0-pre.0/rand/trait.RngCore.html

PR #91 (2019-09-04): Add hal for samd51 TRNG
Closes #27 

Here's some random u32s it generated 
[image]
closed 2019-09-04 by sajattack