atsamd-rs/atsamd #28: Support for ATWINC1500?

zmitchell, 2018-11-28:
Adafruit has a Feather board with an ATWINC1500 WiFi module onboard. I'm wondering what it would take to get a driver written for that chip. I've never written a driver before, but I'm at least considering it. I looked around on the awesome-embedded-rust repo, and there aren't really any other WiFi module drivers listed on that page. It makes me wonder if there's a reason i.e. it's a ton of work, there's a bunch of hidden complexities related to network connections, etc. Does anyone have any insight here?

sajattack, 2018-11-28:
I found a pretty comprehensive programming guide, but it makes the assumption that you're using their provided C libraries. http://ww1.microchip.com/downloads/en/DeviceDoc/ATWINC15x0-Wi-Fi-Network-Controller-Software-Design-Guide-User-Guide-DS00002389B.pdf

sajattack, 2018-11-28:
Also, it's not an ARM part, it's Cortus APS3.

zmitchell, 2018-11-28:
Couple of questions:

* Could we leverage any of smoltcp for the networking bits?
* Which parts are the WiFi module’s firmware responsible for, and which parts would I be responsible for if I’m talking to it from the ATSAMD21?
* Why does the architecture of the WiFi module matter? I assumed I would just be talking to it from the ATSAMD21 and it would already have its own firmware.

wez, 2018-11-28:
I'm not super familiar with the board you mentioned, but taking a quick peek, I can see that adafruit offer a breakout board with that chip that is addressable via SPI:
https://learn.adafruit.com/adafruit-atwinc1500-wifi-module-breakout/overview
The Feather version embeds the same thing and is also addressable via SPI.

In the arduino world, you'd use this library to talk to the device:
https://github.com/arduino-libraries/WiFi101
A lot of that code relates to DNS and TCP and a smallish corner appears to be device specific. (Caveat: I've only very briefly skimmed the filenames in that repo, so I don't have any real knowledge of what's really going on in there).

Based on this, to use this device from rust code you'll need:

* This hal to provide access to the SPI peripheral
* a TCP stack (`smolctp` seems like a good place to start)
* a driver that smoltcp can use (or that can be used with smoltcp) to address the atwinc1500 via SPI. This probably isn't a huge amount of code, but will require some knowledge of how to relay the data between smoltcp and the device.  If you're lucky, there will be a handful of core functions to implement (eg: send a packet, poll for a packet, configure an address/MAC) and then smoltcp can take care of the heavy lifting.  This would make a nice embedded rust module.

Both smoltcp and the driver are technically out of scope wrt. `atsamd-rs` as the code that needs to be written is not specific to ATSAMD; you could use that breakout with any other system that can speak SPI, and that what you should target if you decide to take a crack at this

Once that driver is available, I think it makes a lot of sense to add to this organization a board support crate for that feather board that has a dependency on the driver crate.

closed 2019-12-29 by sajattack