atsamd-rs/atsamd #32: SPI IdleLow does not IdleLow

Rahix, 2019-01-01:
I am currently trying to drive the Circuit Playground's Neopixels using SPI.  For doing so @david-sawatzke started writing an [experimental crate](https://github.com/david-sawatzke/ws2812-spi-rs).  However, while trying to use it, I noticed the first pixel to always display `#008000` instead of `#000000`.  As we were able to determine with a logic-analyzer, the reason is that the SPI is `IdleHigh` and this is being interpreted as the first bit being one (`ws2812` uses green-red-blue).  My code:

```rust
let mut peripherals = Peripherals::take().unwrap();
let core = CorePeripherals::take().unwrap();
let mut clocks = GenericClockController::with_internal_32kosc(
    peripherals.GCLK,
    &mut peripherals.PM,
    &mut peripherals.SYSCTRL,
    &mut peripherals.NVMCTRL,
);
let mut pins = hal::Pins::new(peripherals.PORT);

let spi_pinout = hal::sercom::SPI5Pinout::Dipo2Dopo2 {
    miso: hal::sercom::Sercom5Pad2::pb0(pins.foo, &mut pins.port),
    mosi: hal::sercom::Sercom5Pad3::pb23(pins.neopixel, &mut pins.port),
    sck:  hal::sercom::Sercom5Pad1::pb17(pins.bar, &mut pins.port),
};

let gclk = clocks.gclk0();
let spi = hal::sercom::SPIMaster5::new(
    &clocks.sercom5_core(&gclk).unwrap(),
    3_000_000.hz(),
    embedded_hal::spi::Mode {
        polarity: embedded_hal::spi::Polarity::IdleLow,
        phase: embedded_hal::spi::Phase::CaptureOnFirstTransition,
    },
    peripherals.SERCOM5,
    &mut peripherals.PM,
    spi_pinout,
);

let mut ws = ws2812_spi::Ws2812::new(spi);
```
(The pins `pins.foo` and `pins.bar` are `PB0` and `PB17` respectively, I just added them to get the SPI peripheral working ...)

Is there something wrong on my side or is this an issue in the `hal`?

wez, 2019-01-01:
Just doing a quick drive by; I'm traveling and don't have any hardware to verify things.
`IdleLow` corresponds to setting `CPOL` to `0` (https://github.com/atsamd-rs/atsamd/blob/master/hal/src/sercom/spi.rs#L162) which appears to be the correct setting for this mode based on the datasheet.

I also want to note that, on the Metro M0 and probably also this board, one of the hardware debugger lines is shared by one of the neopixel connections which results in "random" pulses being sent to the on-board neopixel and making it change color and brightness seemingly at random when debugging with a SWD attached.   Is it possible that that is what you're seeing here?  Try executing your build without the debugger attached (remember to disable semihosting debug prints if you happen to have those enabled too).

Rahix, 2019-01-01:
Hmm, I don't think so :/ First of all, I don't have any debugger attached, at least none that I know of ... Also, I was able to control the pixels just fine by manually bitbanging the ws2812 protocol. I will continue to experiment, maybe I can narrow it down a little more ...

david-sawatzke, 2019-01-01:
What we saw is that, while nothing was written, the spi data line was high, thus setting the first bit and shifting all further bits for the other leds.

sajattack, 2019-01-01:
Welcome, thank you for your work. As we discussed on irc, I have also been working towards a similar goal. I'll do my best to help test this issue and improve the hal. Would you be interested in adding your code as a backend for mine? I'm trying to keep the structure similar to Adafruit's, where we have a high level library, and a series of "write" backends. https://github.com/sajattack/ws2812-rs/tree/master/src

david-sawatzke, 2019-01-01:
I'd like to do the 'write' stuff iterator based, but I'll open an issue there.

sajattack, 2019-01-01:
I do also recall having issues with the first neopixel, I'm glad you were able to track down the issue.

Rahix, 2019-01-01:
@sajattack: Well, my issue is only present if I use SPI.  Your code seems to bitbang, I also tried that and did not have this issue there ... Maybe your implementation accidentally leaves the line high when not sending?

sajattack, 2019-01-01:
I have a SPI implementation that I haven't pushed.

sajattack, 2019-01-01:
Possibly the best thing to do is fire up a debugger and compare configuration register values.

sajattack, 2019-01-02:
Could it be high before the SPI initialization starts? What happens if you write a few zeroes at the start of your data? Read this article, it talks about sending a byte of zeroes before the actual data: https://www.rogerclark.net/arduino-stm32-neopixels-ws2812b-using-spi-dma/

david-sawatzke, 2019-01-03:
That may be an issue, but in this case the mosi line was continually high between transfers. We checked with a logic analyzer.

sajattack, 2019-01-17:
Can confirm.

Rahix, 2019-01-22:
Huh, I was wrong ... `IdleLow` refers to the `CLK` pin which in this case is not used so it does not matter.  What I would like to have is the `MOSI` pin being idle low.  Unfortunately this is not easily possible.  I found the following in the Adafruit sources:

> When idle, SPI logic level is normally HIGH, we need LOW for latch.  There is no invert option.  Various tricks like switching the pin to a normal LOW output at end of data don't quite work, there's still small glitches.  So, solution here is to keep the SPI DMA transfer in an endless loop...it actually issues the NeoPixel data over and over again forever (this doesn't cost us anything, since it's 100% DMA, no CPU use)...and those 90 zero bytes at the end provide the 300 microsecond EOD latch.  Hack!

I guess this issue is closed ...

sajattack, 2019-01-23:
Nice digging @Rahix. @wez, does our interpretation of IdleLow (the clock, not the mosi) match the rest of rust-embedded?

Rahix, 2019-01-23:
Yes, rust-embedded does this right: [**Clock** Polarity](https://docs.rs/embedded-hal/0.2.2/embedded_hal/spi/enum.Polarity.html)

I guess, @david-sawatzke and me were just too tired at congress to notice :stuck_out_tongue:

closed 2019-01-23 by sajattack