atsamd-rs/atsamd #39: ATSAMD51 DAC support

tarcieri, 2019-01-22:
This is a tracking issue for supporting the ATSAMD51's DAC. I'm primarily interested in supporting the AdaFruit NeoTrellis M4 (see https://github.com/rust-embedded/wg/issues/286). Wiring it into an ADC abstraction (#35) sounds cool, but for now I'd be happy just getting it working with the [low-level API](https://docs.rs/samd21g18a/0.2.1/samd21g18a/dac/index.html).

The main thing I have to go off of is AdaFruit's port of the Teensy audio engine to these platforms. Here are a few relevant commits:

- "DM: working w/ dual dac only" https://github.com/adafruit/Audio/commit/6d1ee585adb0e39ef34abf6f55544f6e18fc1db4
- "DM: interrupts fix and update synthesis examples for SAMD51" https://github.com/adafruit/Audio/commit/4a8bc7144effb58905f3c445422e14dd928b6516
- "DM: fix DMA interrupt structure for DAC output, other SAMD51 updates" https://github.com/adafruit/Audio/commit/8f4325807503a655e30ef304da7d75c3d89dda6f

Is DMA going to be an issue?

For what it's worth, I'd like to attempt to port this PureData synthesis engine to this platform:

https://github.com/mhroth/ZenGarden

I started a project to do that here:

https://github.com/neobirth/zengarden.rs

This project supports NEON acceleration, although I'm not sure if it's compatible with the subset of NEON (DSP instructions) supported by the ATSAMD51(G19A?). Otherwise it appears to be an easy-to-embed PureData engine that runs inside of an external event loop written in C++ that presents a C interface.

sajattack, 2019-01-24:
The low-level API portion is auto-generated and will be part of my SAMD51 PR.

tgross35, 2024-01-11:
Since there is now ADC support in the HAL https://docs.rs/atsamd-hal/latest/atsamd_hal/adc/index.html and the PAC https://docs.rs/atsamd51g/latest/atsamd51g/adc0/index.html, could this be closed?

ianrrees, 2024-01-11:
I think so!

sajattack, 2024-01-11:
ADC != DAC?

tgross35, 2024-01-11:
Misread, thanks

ianrrees, 2024-01-11:
Whoops, thanks for the catch @sajattack!