atsamd-rs/atsamd #44: Add RTC example, or document peripheral activation.

everett1992, 2019-03-18:
I cannot figure out how to use the RTC. 

> 19.5.3 The RTC bus clock (CLK_RTC_APB) can be enabled and disabled in the Power Manager, and the default state of CLK_RTC_APB can be found in the Peripheral Clock Masking section. A generic clock (GCLK_RTC) is required to clock the RTC. This clock must be configured and enabled in

16.6.2.6 shows that CLK_RTC_APB is enabled by default, and the apbamask shows that is is on.
```rust
peripherals.PM.apbamask.read().rtc_().bit() // true
```

I tried writing RTC enable bit but it blocks forever.
```rust
let rtc = peripherals.RTC.mode0();
rtc.ctrl.write(|w| w.enable().clear_bit()); // <- this blocks forever
```

I know that I need to enable a clock generator, but I don't know how to connect a generator to the RTC peripheral.

---
Thanks for writing this crate, it helped me build [this calculator](
https://hackaday.com/2019/03/17/custom-calculator-rolls-d20-so-you-dont-have-to/?fbclid=IwAR1IYuRLkDdo2XcZw_CfjDKmrROX_7yHBsyun8kYLAFGvnyQC0oytxSEFR4). Now I want to use the RTC to seed the PRNG.

sajattack, 2019-03-18:
Hi everett, I responded to you on irc when you were asking there a week or so ago, but you had already left the channel. Unfortunately, at this time we have not built up a hardware abstraction layer dedicated to the RTC, so it requires diving deep into the dirty register writes, as you have seen. I expect what you're seeing in terms of the blocking is that the peripheral isn't set up properly. So you have two options: figure out all the steps required to initialize the RTC, or find a simpler way to seed your rng.

sajattack, 2019-03-18:
Maybe a good option would be the cortex-m dwt cycle counter? https://docs.rs/cortex-m/0.6.0/cortex_m/peripheral/index.html

everett1992, 2019-03-18:
I'd rather figure out the steps required to initialize the RTC. I want to do more with rust on hardware so it would be beneficial to understand how to create my own abstraction layers. I've read the reference sections on clocks, generic clock controllers, power management, the rtc, and synchronized reads. I'll go back and read the rest of the book.

As far as I can tell there are only abstraction layers built for digital input pins, clock controllers, spi, i2c, and countdown timers. Maybe usb uart based on reading some of the example code. That leaves most peripherals without abstractions. Is there advice you could add to the docs that would help people write abstraction layers or interact with low level hardware?

sajattack, 2019-03-18:
When I'm adding something to the HAL, I usually read the datasheet, an existing C library for the peripheral, and existing sections of the PAC and HAL. Other than that advice it's all pretty specific to what you're working on.

everett1992, 2019-03-18:
Can you recommend a particular c library to reference? Can you link to any
commits or pull requests where you implement a HAL for a peripheral?

sajattack, 2019-03-18:
I found this arduino RTC library for SAMD21: https://github.com/arduino-libraries/RTCZero/blob/master/src/RTCZero.cpp
Pretty much everything under hal/src of this repo would work in terms of looking at a hal for a peripheral.

sajattack, 2019-03-21:
This may or may not be helpful https://www.youtube.com/watch?v=_M0AYdWKnzw A video where someone is setting up an RTC on another microcontroller.

everett1992, 2019-03-21:
Thanks, I'll work on this this weekend

everett1992, 2019-03-28:
I haven't forgotten about this.

I added a line for the RTC [here](https://github.com/atsamd-rs/atsamd/blob/master/hal/src/clock.rs#L313)

and the RTC works if I call

```
clocks.rtc(

sajattack, 2019-03-28:
I'll happily review a pull request.

TDHolmes, 2020-10-22:
I think this is closed by #284?

sajattack, 2020-10-23:
Maybe, but there are still other RTC modes I think.

TDHolmes, 2020-10-26:
#291 adds a driver for RTC in 32 bit timer mode, but only for thumbv6 targets

sajattack, 2020-11-16:
@twitchyliquid64 @TDHolmes Can you guys even out the target support for rtc modes so I can close this issue? Are there any other modes we should worry about?

twitchyliquid64, 2020-11-16:
Yeah I only implemented clock/calendar on M4. Fwict @TDHolmes implemented counter mode on M0? Is counter mode all that needs to be ported across to M4?

sajattack, 2020-11-16:
Yeah. Do you know if there are other modes as well?

twitchyliquid64, 2020-11-16:
The d/e 5x has 16-bit counter, 32-bit counter, and clock/calendar modes. Cannot speak to the d21, but from memory its got similar features but different IP.

I can implement the counter modes on M4 if thats all thats needed on thumbv7em. I don't have d21 hardware readily available (well, theres an itsybitsy_m0 in a box somewhere in my apartment), so would prefer if any work on that front is picked up by someone else.

sajattack, 2020-11-16:
Sounds good.

TDHolmes, 2020-11-24:
I can give porting calendar mode to D21/D11 a crack during this
Thanksgiving weekend. I have a feather M4 now too if counter mode still
needs porting after finishing calendar mode port. I don't feel 16 bit
counter mode is useful with 32 bit counter mode implemented.

TDHolmes, 2020-11-28:
I've implemented a unified RTC driver in #346. let me know what you guys think! @twitchyliquid64 @sajattack

PR #346 (2020-11-28): make a unified RTC driver for all chip types with both timer & clock capabilities
Refactors the existing M4 clock/calendar driver and M0 timer driver into a unified driver for both modes. Tested the timer mode changes on my feather M0 and feather M4 (updated examples) but haven't tested the wio_terminal clock example or the clock/calendar on M0 devices.

- [x] Test the wio_terminal clock example (can someone help with this? @jessebraham?)
- [x] Add a M0 clock/calendar example

should close #44
closed 2020-11-28 by twitchyliquid64