atsamd-rs/atsamd #49: Support for the Arduino MKR1000

ajmwagar, 2019-03-23:
# Arduino MKR 1000 Support

The [Arduino MKR1000](https://store.arduino.cc/usa/arduino-mkr1000) is a Wifi connected board build on SAMD21. It is very similar to the MKRZERO which this crate currently supports.

It would be great if support could be added for the MKR1000 either by extending the `arduino_mkrzero` crate or creating a new `arduino_mkr1000` crate that supports the MKR1000.

Thanks in advance!

Best,

Avery

sajattack, 2019-03-24:
Are you planning to use the Wi-Fi? We don't really have a great network stack set up in the rust ecosystem yet that would make it easy to incorporate afaik.

ajmwagar, 2019-03-24:
WiFi support wouldn't be required. But if we could have a starting point for further implementation, (i.e. a WiFi module) that would be great.

sajattack, 2019-03-24:
I just looked and the pinouts and chip are identical, so you should be able to use the mkrzero crate until you add wi-fi support, at which point we can revisit this.

ajmwagar, 2019-03-25:
Ok, thanks!

Closing for now.

closed 2019-03-25 by ajmwagar