atsamd-rs/atsamd #58: RFC: Should Miso/Mosi and TX/RX be optional?

sajattack, 2019-05-05:
Would like to see what everyone thinks about changing the type to Option, so that you can do one way SPI and UART without wasting pins.

wez, 2019-05-17:
If the hardware supports that then I think its generally best for our interface to expose it.  We just need to balance that with ergonomics; making the 80% use case the most ergonomic seems like a good aspiration, while still making it possible for the 20% use case.

Making a couple of things optional doesn't sound too heavy to me, so it's probably fine.  I suspect that seeing a PR will help polarize/crystallize opinions on this a bit more :-)

BenBergman, 2019-05-18:
I just ran into a case yesterday where I really wished I could leave out the MISO pin of my SPI peripheral. I'm all for seeing this happen.

taurr, 2020-01-08:
All in favor - already had to work around this once, lucky for my project the extra pin could be assigned to a HW pin that was unused.

Check, should be done :-)

bradleyharden, 2020-11-04:
I have this implemented in my `v2` of the SPI module, supported by my `v2` of the SERCOM `pads` module. I hope to get those PRed soon.

However, there's one issue. How do you trigger a transaction for a read-only SPI instance? Presumably the SPI instance would not implement `FullDuplex`. The way I see it, there are three possible solutions. You either need to:

- initiate a transaction on every call to `read` and make it completely blocking
- store some state within the `read` function and use that to determine when to start a new transaction on subsequent calls
- provide a separate function that triggers transactions, like `send` would do in `FullDuplex`

Right now, I've implemented the 3rd approach, with a function name of `trigger`. What do you think? Does anyone have a strong preference or a better idea? @blm768, do you have any thoughts?

blm768, 2020-11-05:
For the sake of flexibility, I think I'd favor option 2 or 3. Option 2 strikes me as being more ergonomic and less likely to get users into a state where they're blocking on a read that'll never happen because they didn't trigger a transaction, but I'm not sure how many edge cases we'd have with the mutable state.

bradleyharden, 2020-11-07:
I agree that option 2 is the most ergonomic, but I worry that it may be possible for the stored state to get out of sync with the hardware state. That would be a disaster. If we're not confident we can prevent that, I would argue against it. Do you have any thoughts?

blm768, 2020-11-08:
I guess we've got similar risks with the second and third APIs; it's just that the third API makes the user responsible for managing that risk. Perhaps we can check the DRE register or something if we're uncertain what the current SPI state is?

bradleyharden, 2021-02-16:
Now that `sercom::v2::spi` has been merged, I think we can close this.

I went with something that looks like option 2 above. `FullDuplex` is implemented when you have both MISO and MOSI. `serial::read` is only implemented when you have MISO but not MOSI (for Master mode). That implementation stores internal state, and I tried to make sure it can't get out of sync using only safe code. `serial::read` is implemented in Slave mode as well, when you only have MOSI, but it doesn't need to store any internal state, because it doesn't initiate transactions.

closed 2021-02-17 by twitchyliquid64