atsamd-rs/atsamd #64: I2S Support

ane, 2019-05-20:
ATSAMD21/51 both support I2S. Is there a plan to add this to the HAL?

sajattack, 2019-05-20:
Yeah, there's a lot of peripherals we don't support yet. Even basic stuff like ADCs and DACs, which should probably take priority over I2S. PRs are always welcome, It won't be on the top of my todo list for quite some time.

ane, 2019-05-21:
I figured as much. I have a MEMS microphone breakout board and a bunch of ATSAMD boards from Adafruit lying around so I'll see if I can write something [like this for the STM32F429](https://docs.rs/crate/stm32f429-hal/0.1.1/source/src/i2s.rs). I also have an analog mic as well so maybe I'll take a look at ADC as well.

sajattack, 2019-05-21:
I'm currently working on ADC here https://github.com/atsamd-rs/atsamd/pull/56

ane, 2019-05-21:
Cool! Are you making good progress with it? 

I guess I'll see what I can come up with regarding I2S.

sajattack, 2019-05-21:
It's done for SAMD51, not started for SAMD21, I've been working on other things.

ianrrees, 2020-11-14:
I've started on the I2S peripheral - the project that prompted the work is for TDM, but the differences from I2S (and maybe PDM) look fairly minor.  https://github.com/ianrrees/samd-i2s

ianrrees, 2021-06-28:
I've moved the I2S work to a branch in my fork [here](https://github.com/ianrrees/atsamd/tree/add-i2s), and have ordered an [I2S DAC board](https://www.adafruit.com/product/3678) and a couple [PDM mics](https://www.adafruit.com/product/3492) to make up a couple examples with.

backtail, 2023-07-08:
Hi, I would be highly interested in I²S support for the ATSAMD51J. If there is any interest in contributing to the HAL instead of me writing a custom solution for my custom problem, I would gladly help and take a look at @ianrrees code.

ianrrees, 2023-07-09:
I'm happy to answer any questions, though my memory is a bit fuzzy and the project that prompted me to work on this seems perpetually stuck in the near future at work...