atsamd-rs/atsamd #69: Upgrade digital input/output pins to embedded_hal::digital::v2

sajattack, 2019-06-28:
As of embedded_hal [0.2.3](https://docs.rs/embedded-hal/0.2.3/embedded_hal/), there is a new fallible (returns a Result) version of the [digital traits](https://docs.rs/embedded-hal/0.2.3/embedded_hal/digital/index.html) and the non-fallible version we are using is deprecated. It looks like there are compatibility shims to upgrade v1 (the version we're using) to v2, but I don't know all the details. As usual, probably a good idea to see what other HALs like [stm32f1xx](https://github.com/stm32-rs/stm32f1xx-hal) and [nrf52](https://github.com/nrf-rs/nrf52-hal) are doing.

sajattack, 2019-06-28:
cc @fionawhim

fionawhim, 2019-06-28:
As far as I can tell we’re pioneers in switching to v2. I don’t _think_ there are error states? So mostly this is just type wrangling.

BenBergman, 2019-06-28:
IIRC, the failing trait was added so that pins on IO expanders could be used interchangeably with direct IO when passed to drivers while still letting drivers fail gracefully if something wasn't right.

sajattack, 2019-06-28:
Hmm, ok. Then why is v1 deprecated?

sajattack, 2019-06-28:
The page for v2 says "Infallible implementations should set Error to !."

fionawhim, 2019-06-28:
@sajattack `!` as a type would force us into nightly from what I can tell. Would `()` be ok for `Error`?

sajattack, 2019-06-28:
I suppose so. Maybe we're jumping into this too early.

fionawhim, 2019-06-28:
Well, I can make the branch and we can see the consequences.

closed 2019-07-09 by sajattack