atsamd-rs/atsamd #78: Arduino/Genuino Zero Support

commonbreed, 2019-07-27:
Are there any plans for [Zero](https://store.arduino.cc/genuino-zero) support?

I see there's MKRZERO support, but I'm unsure regarding how similar the pin assignments are on these boards.

djmcgill, 2019-07-27:
The answer is unfortunately that they are not especially similar, you can compare the two:
- https://github.com/arduino/ArduinoCore-samd/blob/master/variants/mkrzero/variant.cpp
- https://github.com/arduino/ArduinoCore-samd/blob/master/variants/arduino_zero/variant.cpp

However adding the mapping between board pins and chip pins shouldn't be too bad, it would look extremely similar to the [mkrzero one](https://github.com/atsamd-rs/atsamd/blob/master/boards/arduino_mkrzero/src/lib.rs) but with the values from arduino_zero sheet above.

commonbreed, 2019-07-28:
Nice, this doesn't look too difficult to implement. When I have a minute, I'll do it - cheers. I'll leave this open until then.

CrafterSvK, 2020-01-07:
Hello, I was porting Nano 33 Iot and after one upload the USB communication stopped working. There is a space reserved so the bootloader won't be rewritten but somehow it stopped working. I will update you here just in case there is a same problem with Zero.