atsamd-rs/atsamd #83: No UART RX data

mogenson, 2019-08-06:
Hello, I'm debugging some bidirectional UART issues with a simple serial loopback example for a Trinket M0 board. However, I'm unable to read any RX data over UART. The `serial.read()` call blocks forever, maybe because the RXC bit is never set.

I've verified that the Trinket M0 board and USB to UART adapter are working correctly with an Arduino sketch. I'll try to get my hands on a samd21 board with a SWD debug header so I can poke things with gdb. But for now, could someone let me know if I've initialized the UART0 device correctly?

New file `boards/trinket_m0/examples/uart_loopback.rs`, built on top of commit 53f8d846e.
```
#![no_std]
#![no_main]

#[macro_use(block)]
extern crate nb;
extern crate panic_halt;
extern crate trinket_m0 as hal;

use hal::clock::GenericClockController;
use hal::prelude::*;
use hal::{entry, CorePeripherals, Peripherals};

#[entry]
fn main() -> ! {
    let mut peripherals = Peripherals::take().unwrap();
    let mut core = CorePeripherals::take().unwrap();
    let mut clocks = GenericClockController::with_internal_32kosc(
        peripherals.GCLK,
        &mut peripherals.PM,
        &mut peripherals.SYSCTRL,
        &mut peripherals.NVMCTRL,
    );
    let mut pins = hal::Pins::new(peripherals.PORT);
    let mut red_led = pins.d13.into_open_drain_output(&mut pins.port);

    let mut serial = hal::uart(
        &mut clocks,
        9600.hz(),
        peripherals.SERCOM0,
        &mut core.NVIC,
        &mut peripherals.PM,
        pins.d3,
        pins.d4,
        &mut pins.port,
    );

    serial.bwrite_all(b"ready\r\n").unwrap();
    loop {
        let byte = block!(serial.read()).unwrap();
        red_led.toggle();
        block!(serial.write(byte)).unwrap();
    }
}
```

sajattack, 2019-08-06:
I think @keithduncan had a similar issue. I'll do some digging.

sajattack, 2019-08-06:
Try commenting line 190 `w.rxc().set_bit` of hal/src/sercom/uart.rs and let me know if that fixes it.

mogenson, 2019-08-07:
Thanks for the suggestion. The above loopback example works if the INTSET.RXC bit is not set during init.

```
                        // sercom.usart().intenset.modify(|_, w| {
                        //     w.rxc().set_bit()
                        //     //w.txc().set_bit()
                        //     //w.dre().set_bit()
                        // });
```

Does the atsamd crate define a SERCOM0 interrupt handler somewhere I can't find? Or does the first SERCOM0 interrupt go to the cortex-m-rt default handler infinite loop?

How can I define my own interrupt handler? The following didn't work for me.

```
use hal::interrupt;

#[interrupt]
fn SERCOM0() {
}
```

sajattack, 2019-08-07:
Nah, I don't think you need to define an interrupt handler, I think I accidentally turned it on when I shouldn't have.

mogenson, 2019-08-07:
Fixed with 06aa04d

closed 2019-08-07 by mogenson

keithduncan, 2019-08-08:
I found I had to use interrupts due to receives being asynchronous, it will depend on what you’re connecting to on the other end and whether it sends you any unsolicited bytes. You might run in to a receive overrun error if you don’t read the received bytes quick enough.

I’ve copied a serial example I used to test the sercom configuration into a [gist](https://gist.github.com/keithduncan/226117d7dfc81e5f96ba18685b0d2705).

I’m using interrupt reads in my project and am happy to keep this in my fork. Maybe the interrupt mode in use should be an argument to the UART constructor?

keithduncan, 2019-08-08:
I also had to use the external crystal for reliable communication with my modem (`GenericClockController ::with_external_32kosc`).

I also ended up hiding the UART serial interface behind a ring buffer. This made it easier to consume the asynchronous byte stream in the rest of my program and might be worth adding to the hal or another crate.

sajattack, 2019-08-08:
I'd welcome improvements to the hal's uart rx routine.