atsamd-rs/atsamd #84: Add support for Sodaq SARA AFF boards

langestefan, 2019-08-06:
Hi,

I would like to extend Rust support for SARA Arduino Form Factor boards made by @SodaqMoja.
Link: [Sodaq Sara_AFF](https://learn.sodaq.com/Boards/Sara_AFF/)

They feature the ATSAMD21J18. I looked around in the repo but it seems support is not yet available for this particular MCU. They also feature a range of sensors and AT command based cellular modules that support LTE-M and NB-IoT. 

I could try building the necessary files myself but I am not an experienced developer nor do I have any experience with Rust. It would therefore be great if somebody could provide support for this board or guide me in the process.

sajattack, 2019-08-07:
On it.

langestefan, 2019-08-07:
Great!! @sajattack

sajattack, 2019-08-07:
Just to be clear, I'm adding atsamd21j18a, and I'll let you add the board if that's ok.

langestefan, 2019-08-07:
> Just to be clear, I'm adding atsamd21j18a, and I'll let you add the board if that's ok.

That's cool. Do you have any documentation on this? Any process I should follow?

sajattack, 2019-08-07:
https://github.com/atsamd-rs/atsamd/wiki/Adding-a-new-board

langestefan, 2019-08-08:
Everything went fine but I can't seem to get a succesful build of the example: 
```
warning: build failed, waiting for other jobs to finish...
error: could not exec the linker `cc`
  |
  = note: Permission denied (os error 13)
```
Any advise? I'm using Windows Subsystem for Linux (WSL) on Windows 10 btw.

sajattack, 2019-08-08:
Hmm. Do you have gcc installed?

langestefan, 2019-08-08:
Yes. I managed to get it to compile, but I see the following errors: 

```
error[E0432]: unresolved import `crate::target_device::gclk::clkctrl::GENR`
 --> /home/langestefan/atsamd/atsamd/hal/src/clock.rs:6:42
  |
6 | use crate::target_device::gclk::clkctrl::GENR::*;
  |                                          ^^^^ could not find `GENR` in `clkctrl`

error[E0432]: unresolved import `crate::target_device::gclk::clkctrl::IDR`
 --> /home/langestefan/atsamd/atsamd/hal/src/clock.rs:7:42
  |
7 | use crate::target_device::gclk::clkctrl::IDR::*;
  |                                          ^^^ could not find `IDR` in `clkctrl`

error[E0432]: unresolved import `crate::target_device::gclk::genctrl::SRCR`
 --> /home/langestefan/atsamd/atsamd/hal/src/clock.rs:8:42
  |
8 | use crate::target_device::gclk::genctrl::SRCR::*;
  |                                          ^^^^ could not find `SRCR` in `genctrl`
```

And a bunch more errors for other values.

```
Some errors have detailed explanations: E0408, E0412, E0425, E0432.
For more information about an error, try `rustc --explain E0408`.
error: Could not compile `atsamd-hal`.
```

I will try my best to fix it...

sajattack, 2019-08-08:
Are you compiling from the board directory or the hal directory?

langestefan, 2019-08-08:
Board directory. I am following the instructions in the readme

sajattack, 2019-08-08:
It could be that the recent svd2rust changes caused an issue.

langestefan, 2019-08-08:
I'm not sure. Is there any way to figure that out? Can we try again with a previous version?

sajattack, 2019-08-08:
Yeah so it looks like IDR has changed to ID_A and SRCR to SRC_A. I'll see what I can do.

langestefan, 2019-08-08:
Cool. Do you need the full error log? Because there were a number of other identical errors but for different modules.

sajattack, 2019-08-08:
I just reverted svd2rust.

sajattack, 2019-08-08:
Broke everything oopsie

langestefan, 2019-08-08:
I noticed... aw well

sajattack, 2019-08-08:
Latest commit looks better.

langestefan, 2019-08-08:
I had to change:

```
[dependencies.atsamd-hal]
version = "~0.6"
```
To:
```
[dependencies.atsamd-hal]
version = "~0.5"
```
Before it would compile. However, now I am getting the following error when executing cargo build --examples:

```
error: no rules expected the token `#`
   --> src/lib.rs:122:5
    |
122 |     /// GPS_TIMEPULSE
    |     ^^^^^^^^^^^^^^^^^ no rules expected this token in macro call
```

sajattack, 2019-08-08:
Can you share your lib.rs?

langestefan, 2019-08-08:
See https://github.com/langestefan/atsamd/blob/master/boards/sodaq_sara_aff/src/lib.rs

langestefan, 2019-08-08:
I already see what's wrong. I'm missing a comma...

langestefan, 2019-08-08:
Compiling is now OK.

langestefan, 2019-08-08:
Okay so next challenge. How do I actually upload code to the board? WSL is pretty limited when it comes to tooling, installing arduino is not an option and I do not have J-Link.

sajattack, 2019-08-08:
Probably bossac

langestefan, 2019-08-09:
https://launchpad.net/ubuntu/trusty/amd64/bossa-cli/1.3~20120408-3 this should work right?

sajattack, 2019-08-09:
Looks pretty old but maybe. Best way to find out is try.

langestefan, 2019-08-09:
Made some progress.
`stty -F /dev/ttyACM1 ospeed 1200` puts the board into bootloader mode but reports not all actions were able to complete successfully, whatever that means.

`bossac` reports there is no device on the serial port, so it's unable to flash anything. As I said before, WSL is very limited so my next step is to get a native Linux PC since running this stuff from a VM has also not been a very good experience. Any way to get this stuff working from Windows 10?

sajattack, 2019-08-09:
You could try the windows version of bossac https://github.com/shumatech/BOSSA/releases/tag/1.9.1

langestefan, 2019-08-09:
Tried various versions, they all crash as soon as I select a serial port.

sajattack, 2019-08-09:
I don't know the specifics of their bootloader but if it's an
arduino-compatible samd21 it should use BOSSA. Try `bossac -e -w -v -b -R
-o 0x2000 <foo.bin>`

langestefan, 2019-08-09:
With a bit of fiddling I got it working! Blinky verified! Shall I create a PR?

sajattack, 2019-08-09:
Yes please :)

langestefan, 2019-08-09:
Thanks for the help!

PR #86 (2019-08-12): Add support for Sodaq SARA AFF boards.
Closes #84
closed 2019-08-12 by sajattack