atsamd-rs/atsamd #87: Problems using trinket_m0 crate in project

jessebraham, 2019-08-10:
Having a bit of a strange problem here. I also have no idea what I'm doing so hopefully I'm just missing something silly.

I was able to build the `blinky_basic` example from the repository for the Trinket M0 no problem by following the instructions in the readme. So far so good.

I then created a new project using `cargo new`, added `trinket_m0 = "0.3.0"` and `panic-halt = "0.2.0"` to my dependencies, and copied the [blinky_basic.rs](https://github.com/atsamd-rs/atsamd/blob/master/boards/trinket_m0/examples/blinky_basic.rs) file verbatim into my `main.rs`. 

Running `cargo build`, I'm presented with the following warnings and errors (duplicated for both `set_high()` and `set_low()`, omitting the latter for brevity):

> warning: use of deprecated item 'hal::prelude::_embedded_hal_digital_OutputPin::set_high': Deprecated because the methods cannot return errors. Users should use the traits in digital::v2.
  --> src\main.rs:30:17
     |
30 |         red_led.set_high().unwrap();
     |                 ^^^^^^^^
     |
   = note: #[warn(deprecated)] on by default
>
> error[E0599]: no method named `unwrap` found for type `()` in the current scope
  --> src\main.rs:30:28
   |
30 |         red_led.set_high().unwrap();
   |                            ^^^^^^

Anybody have any idea what I'm missing here?

sajattack, 2019-08-10:
It's probably been a while since we updated crates.io so the example in the repo might not work with the trinket_m0 crate that's been deployed.

jessebraham, 2019-08-10:
Gotcha. Checked the history on the `blink_basic.rs` file and managed to get it to build, thanks!

closed 2019-08-10 by jessebraham