atsamd-rs/atsamd #88: SAMD51 PAC's use incorrect register offsets for tcc0::CCBUF_DITH4

proman21, 2019-08-30:
There appears to be a mismatch between the TCC.CCBUF register definition found in the D5x/E5x datasheet (Revision E) and what is generated for SAMD51x family PACs.

In the case of the DITH4 resolution, the CCBUFn.CCBUF should occupy bits [23:4], and the CCBUFn.DITHERBUF occupies bits [3:0]. However, the PAC says that CCBUFn.CCBUF occupies [3:0] and the CCBUFn.DITHERBUF occupies [23:4]. The field offset and size has been swapped for CCBUF and DITHERBUF.

This seems to be an error present in the SVD files used to generate this crate, so that will have to be corrected for the PAC to generate correctly. So far I've checked SAMD51J19A and SAMD51G18A SVD's and they seem to both have the error.

sajattack, 2019-08-30:
I blame Atmel/Microchip XD

proman21, 2019-08-30:
> I blame Atmel/Microchip XD

I have and I will. B)

> I have to regenerate all the PACs and then fix the HAL for another reason so it'll probably be done > with that set of changes.

Cool and good. What's the ETA on those changes?

sajattack, 2019-08-30:
When I get to it :stuck_out_tongue:

sajattack, 2019-08-30:
It's fixed on my fork, but it might be a little while before it gets merged because we wanna squeeze in some more hal reorganizing.

proman21, 2019-08-31:
Love it, I was debating doing the same and fixing on a fork but you beat me to it! :confetti_ball: Keep up the great work!

closed 2019-09-04 by sajattack