atsamd-rs/atsamd #94: Peripherals and CorePeripherals not found in root of feather_m0 on v0.5.0

TDHolmes, 2019-09-15:
This code fails to compile on `feather_m0` `0.5.0`, while it worked on `0.4.0`:

```rust
#![no_std]
#![no_main]

extern crate panic_halt;  // panic handler

use feather_m0 as hal;
use hal::clock::GenericClockController;
use hal::delay::Delay;
use hal::prelude::*;
use hal::{entry, Peripherals, CorePeripherals};
```

With this error:

```rust
error[E0432]: unresolved imports `hal::Peripherals`, `hal::CorePeripherals`
  --> src/main.rs:18:18
   |
18 | use hal::{entry, Peripherals, CorePeripherals};
   |                  ^^^^^^^^^^^  ^^^^^^^^^^^^^^^ no `CorePeripherals` in the root
   |                  |
   |                  no `Peripherals` in the root
```

Is this expected? I'm not sure how to remedy this error.

sajattack, 2019-09-15:
Yeah they're in hal::pac now.

TDHolmes, 2019-09-16:
I literally just re-found the examples folder..🤦🏻‍♂️ sorry for the noise. Thanks for a great crate!

closed 2019-09-16 by TDHolmes