atsamd-rs/atsamd #96: Clean up some of feature flag `unproven` ?

twitchyliquid64, 2019-09-19:
Heya,

In trying to get the GPIO's passed through for the bitbang'ed dotstar LED I ran into issues converting the embedded_hal v1 GPIO's into v2 GPIOs. I could only get it to work with the `unproven` feature flag.

I think this is the culprit:

https://github.com/atsamd-rs/atsamd/blob/61e3ee106d1fe20b184a209927cd951304b0973d/hal/src/common/gpio.rs#L339

```rust

#[cfg(feature = "unproven")]
impl<MODE> InputPin for $PinType<Input<MODE>> {
  // TODO: switch to ! when it’s stable
```

v2 seems well established. Can we pull this out of the feature flag?

sajattack, 2019-09-19:
Check if it's still unproven in embedded-hal

twitchyliquid64, 2019-09-19:
Looks like it :sob: 

https://github.com/rust-embedded/embedded-hal/blob/a5c0e3c63e4e431807d491d35dd67d1d43c07c92/src/digital/v2.rs#L129

Guess that means we must have `unproven` there?

sajattack, 2019-09-19:
Yes

sajattack, 2019-09-19:
Do this in .travis.yml https://github.com/atsamd-rs/atsamd/blob/0f7228fd9700c532524c8be9b25598cd9bfbcd1e/.travis.yml#L11

closed 2019-09-19 by sajattack

sajattack, 2019-09-19:
And this in Cargo.toml 
https://github.com/atsamd-rs/atsamd/blob/0f7228fd9700c532524c8be9b25598cd9bfbcd1e/boards/trellis_m4/Cargo.toml#L82