atsamd-rs/atsamd #100: SpinTimer is missing for consumers of itsybitsy_m4?

twitchyliquid64, 2019-10-11:
```
[xxx@xxx]:~/projects/cck> ./cck-dev build
   Compiling cck v0.2.0 (/home/xxx/projects/cck)
error[E0433]: failed to resolve: use of undeclared type or module `SpinTimer`
  --> src/main.rs:33:70
   |
33 |     let mut rgb = hal::dotstar_bitbang(pins.dotstar, &mut pins.port, SpinTimer::new(20));
   |                                                                      ^^^^^^^^^ use of undeclared type or module `SpinTimer`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0433`.
error: Could not compile `cck`.

To learn more, run the command again with --verbose.
```

sajattack pushed new crates, so I'm guessing we need to bump the version in the itsybitsy crate to `>=0.7.1` ?

https://github.com/atsamd-rs/atsamd/blob/b9bc7abf624e01124b9a5576dc06572e2806ea24/boards/itsybitsy_m4/Cargo.toml#L31

sajattack, 2019-10-11:
`cargo update`

twitchyliquid64, 2019-10-11:
FML

Oh well, at least learning is wholesome. Thanks and sorry for the bother!

closed 2019-10-11 by twitchyliquid64

sajattack, 2019-10-11:
Maybe getting rid of the tilde would make this easier idk.