atsamd-rs/atsamd #101: SAMD21 ADC returning values 2x more than expected, except for fully saturated

TDHolmes, 2019-10-14:
I'm using a feather_m0 board and the [AD8495](https://www.adafruit.com/product/1778) thermocouple amplifier. The ADC HAL seems to be working great, except that it returns values exactly 2x of what is expected.

I've tried debugging this, but haven't gotten far. The gain settings are being written in the HAL code, and I don't see any other obvious ways we can get a 2x gain error. I'm reading from pin A5 for the record (pin B2 it seems).

Here's a trimmed down version of my code:

```rust
#![no_std]
#![no_main]

const ADC_FULLSCALE: u32 = 4095; // 12 bit ADC

extern crate cortex_m;
extern crate panic_halt; // panic handler

use feather_m0 as hal;
use hal::adc::Adc;
use hal::clock::GenericClockController;
use hal::delay::Delay;
use hal::entry;
use hal::pac::{CorePeripherals, Peripherals};
use hal::prelude::*;

#[entry]
fn main() -> ! {
    // cortex_m::Peripherals::take().unwrap()
    let mut peripherals = Peripherals::take().unwrap();
    let core = CorePeripherals::take().unwrap();
    let mut clocks = GenericClockController::with_external_32kosc(
        peripherals.GCLK,
        &mut peripherals.PM,
        &mut peripherals.SYSCTRL,
        &mut peripherals.NVMCTRL,
    );

    let mut pins = hal::Pins::new(peripherals.PORT);

    let mut i2c = hal::i2c_master(
        &mut clocks,
        400.khz(),
        peripherals.SERCOM3,
        &mut peripherals.PM,
        pins.sda,
        pins.scl,
        &mut pins.port,
    );  // used for a display

    let mut adc = Adc::adc(peripherals.ADC, &mut peripherals.PM, &mut clocks);
    let mut a5 = pins.a5.into_function_b(&mut pins.port);

    loop {
        let data: u16 = adc.read(&mut a5).unwrap();
        let data: u16 = data / 2;
        let temp_c: f32 = (((data as f32 / ADC_FULLSCALE as f32) * 3.3) - 1.25) / 0.005;
        let temp: f32 = temp_c * (9. / 5.) + 32.;  // print to display
        delay.delay_ms(1000 as u32);
    }
}
```

TDHolmes, 2019-10-14:
Oh I think I figured it out. We're using `intvcc1` reference, and according to this table:
[image]
that means we're 1/2 `VDDANA`, which would explain my 2x discrepancy. This is confusing and opaque... is there something we can do about it? We can't assume something routed to `VREFA`/`VREFB`, but perhaps we should allow this to be selected in initialization?

sajattack, 2019-10-14:
Try my PR and see if it fixes things.

sajattack, 2019-10-14:
As you say, it is confusing and opaque, I had a hard time deciphering the code I was working from and that's probably why this bug crept in. I'll consider making it user-selectable.

TDHolmes, 2019-10-14:
That change will fix things, but it also leaves the 3v3/2 -> 3v3 range of voltages unreadable. I just hacked up a change to make it user selectable, what do you think?

```rust
// (...)
pub use crate::target_device::adc::refctrl::REFSEL_A as refsel;

pub struct Adc<ADC> {
    adc: ADC,
}

impl Adc<ADC> {
    pub fn adc(
        adc: ADC,
        pm: &mut PM,
        clocks: &mut GenericClockController,
        reference: refsel,
    ) -> Self {
        // (...)
        adc.inputctrl.modify(|_, w| w.gain()._1x());
        while adc.status.read().syncbusy().bit_is_set() {}
        match reference {
            refsel::INT1V => adc.refctrl.modify(|_, w| w.refsel().int1v()),
            refsel::INTVCC0 => adc.refctrl.modify(|_, w| w.refsel().intvcc0()),
            refsel::INTVCC1 => adc.refctrl.modify(|_, w| w.refsel().intvcc1()),
            refsel::AREFA => adc.refctrl.modify(|_, w| w.refsel().arefa()),
            refsel::AREFB => adc.refctrl.modify(|_, w| w.refsel().arefb()),
        }
        while adc.status.read().syncbusy().bit_is_set() {}
        Self { adc }
    }
// (...)
```

When I short 3v3 to AREF, I don't get the full range as expected, so there might be something else going on as well?

sajattack, 2019-10-14:
I think how I would prefer to do it would be to default to 3v3 in the constructor, and then have a method that changes it later. We could do the same for sampling rate and other parameters.

commit 635deeee (2019-10-14): correct adc gain (fixes #101) (#102)
closed 2019-10-14 by sajattack