atsamd-rs/atsamd #105: atsamd usb serial not receiving characters until close

jacobrosenthal, 2019-10-20:
I have a usb commit in #104 here for pygamer When you open the port with `screen /dev/blah 115200` and type nothing is printed on screen and Ive checked and the characters arent even received. I thought it was maybe interrupts, but I implemented the polling to upper case echo example and it has the same behavior. 

So maybe my implementation is broken, but wait I noticed as you close screen, the led example will trigger and In the case of the polling to upper case example when you reopen screen you'll see your previous chars printed to screen.

So the act of closing is triggering for some reason. I can't confirm on itsy but I presume its the same. So maybe its my setup for some reason. I checked with [stm32-usb](https://github.com/stm32-rs/stm32-usbd) and a [upper case polling bluepill example](https://github.com/stm32-rs/stm32-usbd-examples/tree/master/example-stm32f103c8
) and can confirm those characters are echoed immediately using screen so its not my setup.

I have a feeling using the stty as given in directions in Itsy, opens and closes and is thus triggering a flush somehow and so this hasnt been noticed until now.

Im not yet familiar with the usb device implementation but I think it would lie in there somewhere. Any thoughts @sajattack ?

twitchyliquid64, 2019-10-21:
Works fine on the itsybitsy_m4, I only implemented 2 weeks or so ago.

> but wait I noticed as you close screen

... Laptop screen?

> I have a feeling using the stty as given in directions in Itsy, opens and closes and is thus triggering a flush somehow and so this hasnt been noticed until now.

One thing that might be worth doing is setting up `usbmon` and having a look at the USB packets going to and from the device in the working / not-working scenarios.

jacobrosenthal, 2019-10-21:
Can you confirm itsy works with screen, not stty

twitchyliquid64, 2019-10-21:
Will do when I get back to my hotel room.

If stty works but not screen, I expect it's due to the CDC ACM driver rather than the peripheral driver. Either way I'll give it a go.

twitchyliquid64, 2019-10-22:
I couldnt get it working at all with screen. I used minicom instead as that _should_ work.

I used wireshark and the `usbmon` module to monitor the USB packets coming out of my device. This is my observations:

 * Setting `stty` thingy then doing the echo thing causes the usb controller to set the line state before and after sending data (this is the scenario that communications work)
 * minicom does not send any control packets except at the beginning
 * Sending data packets only (ie: minicom) hangs interaction till a control packet is sent
 * Hanging up or closing minicom causes the interactions to miraculously start working again (presumably because it results in a control packet being sent).

As such, I think the likely culprit is a code path expecting a control packet and not handling the received non-control packet until that happens.

twitchyliquid64, 2019-10-22:
Either way, this is a valid bug somewhere in the stack (possibly in the usb-device crate, possibly in the peripheral driver, possibly in the usbd-serial crate).

I'll be able to better investigate when I get home (travelling atm), where I can get the UART out and start pumping debug messages.

cc mvirkkunen@ - does usbd-serial work with minicom et al on the devices you tested?

jacobrosenthal, 2019-10-22:
mentioned above, the stm32-usbd on a bluepill example does work as expected
with screen so I dont think its in the upstream traits or on my system. I
think its in the atsamd impl

jacobrosenthal, 2019-10-25:
From @mvirkkunen on matrix
> I haven't exactly tested on anything except one device but "picocom" works on Linux and I forgot which terminal I used on Windows but it worked

jacobrosenthal, 2020-01-15:
Ive got my debugger up and working to look at this. Nothing obvious yet but Ive never dealt with usb.

it seems like trcpt1 isnt being called to trigger the poll when the char is received
instead when disconnect happens, usb_other triggers the poll, which triggers something internally to finally trigger trcpt1 to be hit, forcing another poll, which finally receives the character

twitchyliquid64, 2020-01-16:
There may be a quirk / undocumented behaviour where certain endpoints or states need to be serviced first before other interrupts fire.

A few ideas:

 - Maybe we still need to check each endpoint if we get a suspend or reset interrupt (ie: device doesnt send trcpt1 if there is already a different interrupt pending).
 - Maybe we need to check endpoints _before_ checking suspend / reset interrupts.

jacobrosenthal, 2020-01-17:
I dont think its poll ordering. I wired up the start of frame interrupt, which seems to get fired on probably on every packet including heartbeats or something, to the poll function, and with that hitting poll constantly the reads come through fine.

So the register is showing the buffer has data and clearing it fine. So it does seem that for some reason the trcpt0 (I said trcpt1 last time, I think I now understand that input is on channel 0) interrupt isnt firing to trigger the poll

I looked through svds and those seem correct around interrupts registers...

twitchyliquid64, 2020-01-19:
I spent hours and couldn't figure out why the interrupt wasn't firing. Given I havent found any open-source implementation not using the SOF interrupt, ive just submitted a fix PR that uses SOF. Can confirm it fixes our issues.

jacobrosenthal, 2020-01-19:
Ive come to the same conclusion on suspend/resume

The fact is this chip does reset/suspend/resume and more for you. We could presumably be doing more low power stuff, but were not anyway.

Ive spent time rewriting but I still havent gotten the mix of flags. I THINK you need enable the wakeup interrupt when you suspend? But Ive found that damn bit set constantly and havent been able to then distinguish from suspend and not awake and suspend and awake. Would we need a local static variable?

The question comes down to does usb-device heavily profit from suspend/resume notifications? I dont know...

And yeah for interrupt. Ive spend days now. I just dont see anything in other implementations either

Maybe Ill try to disable sof in the arduino and see if it still works?

PR #154 (2020-01-21): samd51 UsbBus: Fix interrupts not enabling & missing poll events around USB reset
Fixes #105.

So it looks like the issue came down to our writes to `EPINTENSET[n]` not sticking. I found one really important sentence:

> This [EPINTENSET] register is cleared by USB reset or when EPEN[n] is zero.

-- Page 1164 / 2129 of the SAM D5x/E5x Family Data Sheet

There is no register or other reference to `EPEN`. I've taken it to mean a non-zero value for `EPCFG[n].EPTYPE[0/1]`, but to be sure we configure the endpoint interrupts last.

The remaining changes are the result of an iterative audit of regis [... cut]
commit 1badb469 (2020-01-21): samd51 UsbBus: Fix interrupts not enabling & missing poll events around USB reset (#154)

* samd51 UsbBus: Fix interrupts not enabling & missing events around bus reset

Fixes #105.

* Only clear endpoint _bank ready_ bit during configuration/protocol-reset

* Pair programmin jammin (#1)

* Cleanup ep init logic & improve comments.

* Bump HAL version

* cleanup rxstp interrupts

* Remove unnecessary handling of trfail status in poll
commit 32f37d57 (2020-01-23): samd21 UsbBus: Fix implementation & get itsybitsy_m0 example working (#155)

* samd51 UsbBus: Fix interrupts not enabling & missing events around bus reset

Fixes #105.

* Only clear endpoint _bank ready_ bit during configuration/protocol-reset

* Pair programmin jammin (#1)

* Cleanup ep init logic & improve comments.

* Bump HAL version

* cleanup rxstp interrupts

* Remove unnecessary handling of trfail status in poll

* WIP getting USB working on samd21

* samd21 UsbBus: Fix implementation & get itsybitsy_m0 example working

* Bump crate versions

* Remove dangling reference to uart_debug
closed 2020-01-21 by sajattack