atsamd-rs/atsamd #108: Interest in additional device support?

jessebraham, 2019-11-09:
I've been playing with the [ATSAMD11C14A](https://www.microchip.com/wwwproducts/en/ATSAMD11C14) and decided to take a swing at adding HAL support for this device.

https://github.com/jessebraham/atsamd11-rs

I copied the `hal/src/common/` directory verbatim to my project, then modified the `hal/src/samd21` directory to reflect the new `samd11` device. Everything seems to build except the USB module, which I can't get to build in this project either for some reason, so that has been omitted for the time being.

All I've tested on hardware so far is a basic blinky demo I hacked together, but that seems to be working; I plan on doing some further testing.

Any interest in integrating this into the project, or was the goal to stay focused on the `samd21`/`samd51`?

sajattack, 2019-11-09:
We'd be very happy to support more devices.

jessebraham, 2019-11-09:
Awesome, I'll get a PR going. Thanks!

closed 2019-11-09 by jessebraham