atsamd-rs/atsamd #116: Trouble compiling usb_serial example for metro_m0

pragun, 2019-12-12:
Hi, I'm a rust beginner so it is very probable that I am doing something silly,  but on 
`cargo build --example usb_serial`

I get 

` Compiling atsamd-hal v0.7.4 (/home/pragun/rust_experimentation/atsamd/hal)
error[E0433]: failed to resolve: could not find `usb` in `hal`
 --> examples/usb_serial.rs:9:15
  |
9 | use hal::hal::usb::usb_device::bus::UsbBusWrapper;
  |               ^^^ could not find `usb` in `hal`
`
which of course brings up a host of other compile errors. 
Thanks!

jessebraham, 2019-12-12:
It's not just you, unfortunately the HAL for USB isn't in a great state currently. See #11 for more information.

pragun, 2019-12-14:
Ah thanks!

closed 2019-12-14 by pragun