atsamd-rs/atsamd #121: ATSAMD11 PWM pins?

wyager, 2019-12-15:
Hello all, 

I'm not sure if I'm missing something, but I'm having trouble setting up multiple PWM pins on an ATSAMD11C14. I need a total of 4 PWM pins. Would this require adding support for TC2, or maybe TCC0?

sajattack, 2019-12-15:
@jessebraham

jessebraham, 2019-12-16:
Just got back from out of town, I'll take a look at it in the next couple days here.

jessebraham, 2019-12-19:
I've added (and tested) `Pwm2` to my fork (36afab5bac753b26dd8af4ad7cd5cc1d520e0225) which makes use of `TC2`.

As far as using multiple pins, the datasheet seems to imply it's possible:
> Timer/Counter -> 24.8 Signal Description
>
> Refer to “I/O Multiplexing and Considerations” on page 13 for details on the pin mapping for this peripheral. **One signal can be mapped on several pins.**

I believe it is possible to use `TCC0` as well. However, unless I'm mistaken, in both cases our `pwm!` macro needs to be modified as it currently expects a single `TCxPinout` as well as a `TCx` parameter.

@sajattack , am I on track here? The `pinout` member in the `$TYPE` struct does seem kind of like magic and doesn't appear to be read anywhere yet is still working as expected, so I'm clearly missing something there.

wyager, 2019-12-19:
> One signal can be mapped on several pins.

The thing I'm aiming for is to have 4 separate PWM counters. From a not-too-careful reading of the datasheet, it seems like this might be possible with TC{1,2}/WO{0,1}. I might be misunderstanding the way the WO channels work - the datasheet says `Waveform output channels for TC : 2`, whereas it says `Waveform output channels per TCC : 8`. I'm not sure if this means 2 WOs per TC or 2WOs total shared across both TCs. If it's the latter, I think I would have to use TCC0.

sajattack, 2019-12-19:
The mode we use the WO in uses 2 channels for one pwm output.

wyager, 2019-12-20:
One thing that's confusing me about the rust API is that the definition of `atsamd11c14a::tcc0` seems to be missing some stuff compared to what's mentioned in the datasheet/C snippets. E.g. it seems like `cc` should have some additional structure to account for the fact that there are 4x compare channels. All of this stuff is autogenerated right?

sajattack, 2019-12-20:
Could've been missing from the SVD

wyager, 2019-12-21:
Got it, thanks. So what's the best way forward here? Manually edit the SVD file? Ask the manufacturer? Just give up and write everything in C?

wyager, 2019-12-21:
FWIW, I was able to get this working for my needs via the C FFI. So no longer any urgent desire on my end. Still probably worth fixing though.

sajattack, 2019-12-21:
I don't think our HAL supports TCCs yet anyhow so we can worry about it later.

jessebraham, 2019-12-21:
Sorry, had a busy week. Glad you got a solution figured out.

I have started trying to adapt the `pwm!` macro to use `TCC0` instead of `TCx`, just as a proof-of-concept. If/when I get it working maybe there can be some discussion on how to move forward.

sajattack, 2019-12-22:
Can this be closed now?

wyager, 2019-12-23:
Sure, although Jesse, let me know if you want me to reopen so you can track adding TCC0 support here

closed 2019-12-23 by wyager