atsamd-rs/atsamd #126: ATSAME54 support

mvniekerk, 2019-12-22:
I see the repo says "atsamd", but can't find something similar for the Microchip ATSAME54P20A - the chip I'm targeting.

Will this be fine place to submit any chip/HAL support for this MCU?

jessebraham, 2019-12-22:
Currently there is limited device support, more contributions are welcome. There are a couple pages in the wiki concerning [adding a new chip](https://github.com/atsamd-rs/atsamd/wiki/Adding-a-new-chip) and [adding a new board](https://github.com/atsamd-rs/atsamd/wiki/Adding-a-new-board), just for reference.

mvniekerk, 2019-12-22:
Ok cool so contributions for atsame is fine then?

Thanks for the links, they are cool references and well though out.
I have started on a fork - https://github.com/mvniekerk/atsame
SVD loaded and compiling afaict.

jessebraham, 2019-12-22:
I'm assuming so (I don't see any issue with it personally), @sajattack can confirm.

sajattack, 2019-12-22:
It does present a bit of a naming challege. We've already renamed to atsamd-rs from atsamd21-rs as we grew into 51. That wasn't the smoothest transition either. Also, if we were to rename again, what would we even call it? atsam-rs is too general because there are many different sams that are quite different to the ones we're doing, like atsam3x I believe. I'm aware atsame54 is quite close to atsamd51, so I think this change is fine, it's just a bit annoying on the name front. If you're ok with your e stuff being in our d repo, I don't really have much problem with it.

mvniekerk, 2019-12-22:
Seems fair - hardest thing in programming is to name the thing :-)
Yeah the atsamd51 is close to what I'm busy with. Will submit a PR when I can flash an LED.

closed 2019-12-22 by mvniekerk