atsamd-rs/atsamd #137: Error building hal

ooxi, 2020-01-01:
Wanted to try out the metro_m0 examples, but compiling fails:

```
error[E0432]: unresolved import `crate::hal::Pwm`
 --> /atsamd-rs/hal/src/samd21/pwm.rs:2:18
  |
2 | use crate::hal::{Pwm, PwmPin};
  |                  ^^^ no `Pwm` in the root

error: aborting due to previous error

For more information about this error, try `rustc --explain E0432`.
error: could not compile `atsamd-hal`.
```

I'm using the `gcc-arm-embedded` from the ppa and rust 1.40 as described in the README

jessebraham, 2020-01-01:
Adding `--features=unproven` should fix that particular error. I believe there are currently issues with the `rando.rs` and `usb_serial.rs` examples for the Metro M0, but you can build the remaining examples:

`cargo build --example=blinky_basic --example=blinky_rtfm --features=unproven`

ooxi, 2020-01-01:
Thanks, that worked!

closed 2020-01-01 by ooxi