atsamd-rs/atsamd #139: Setting a pin as OpenDrainOutput makes it go low!

taurr, 2020-01-05:
If a pin is not set High before being switched from Input to OpenDrainOutput, it will default to Low.

This is unexpected behavior.

If a pin is being used as e.g. a interrupt trigger output on a common rail, the switch is enough to trigger the external interrupt during startup!

Either the `pin.into_open_drain_output(&mut port);` should set the pin high before switching, or we should have an extra method with the option for doing so!

sajattack, 2020-01-08:
I've reviewed the code in this area and it doesn't look like we're directly instructing it to go low. Can you check if the same thing happens in an Atmel Studio C project?

taurr, 2020-01-08:
It does - it's just part of the HW.

Don't know if *all* Cortex-M0's does it, but at least the entire ATSAMD20 and 21 series does. In order to predict the state after being turned on as an output, it needs to be set beforehand.

For now, the workaround in my application, using pin PA10, is kinda ugly but works:
```
unsafe {
        // HACK: Hack in order to avoid the interrupt pin from going low when
        // transitioning to a open drain output
        (*PORT::ptr()).outset0.write(|bits| bits.bits(1 << 10));
    }
    let interrupt_pin = pins.int.into_open_drain_output(&mut pins.port);
```
Even if this is the official way, we should at least have it documented, and have an easier (non-hard-coded) way of doing it.

sajattack, 2020-01-08:
I don't think it's the HAL's job to special case something like this that is default behaviour of the hardware.

closed 2020-01-08 by sajattack

taurr, 2020-01-08:
@sajattack: I do not agree on this.

When setting a GPIO as Open Drain output, it's state will be defined by the output register - true, BUT initially it is set as low, and the whole point of having a Open Drain Output is to NOT make the output low initially.
The only way to use open drain as it is, is to switch between Input and OpenDrainOutput, not by switching high/low.

This is not an edge case as I see it as I have yet to meet someone that doesn't set the output to a well known state before switching it on.

Please reconsider the close - at least we should have a `open_drain_output_with_state(state: bool)` method on the GPIO pin.