atsamd-rs/atsamd #140: Multiple names for one PIN

CrafterSvK, 2020-01-06:
Hello there, I really like this project and I am working on description of Arduino Nano 33 IOT. I am new to Rust world, but there is not implemented Copy trait on pins. Is there any way to actually have multiple names for one PIN, as some pins are digital and also TX. It is just for abstraction.

sajattack, 2020-01-08:
I've talked to a few people about this and I don't think there's a good, type-safe way to have multiple pins with the same name.

CrafterSvK, 2020-01-08:
So what about pin usb_n on arduino nano 33 iot which is in conflict with one of the digital pins. :/

sajattack, 2020-01-08:
I think the usb_n takes precedence, all pins are digital outputs. Leave a comment saying it's both.

keithduncan, 2020-01-08:
The pin 'names' and pin 'roles' are different.

The Arduino Nano 33 IoT uses the SAMD21G18A which we have support for. In the D21x datasheet, Pa24 can be multiplexed in various different ways. By default Pa24 is a GPIO unless you configure the `PORT` peripheral to multiplex it to the USB peripheral. To make it USB DM you would configure it for function G.

The board support crates assign 'names' to the pins based on how the board has wired them but they can still be multiplexed to different peripherals. For example you could multiplex some of the board's 'digital outputs' into a sercom peripheral for additional UART/SPI/I2C communication, or some leave some of of the analogue inputs un-multiplexed and use them as digital GPIO pins.

CrafterSvK, 2020-01-08:
Okay I understand. I will only label the Arduino pins and everything else is just on the programmer. I will remove usb_p as usb_n is default GPIO.

closed 2020-01-09 by keithduncan