atsamd-rs/atsamd #147: SAMD5x/E5x PACs missing register field TC.CTRLA.DMAOS

proman21, 2020-01-14:
Another SVD mismatch with the reference manual.

Pretty much what it says on the tin. The `DMAOS` field of the `CTRLA` register for TC peripherals is not present in the SVD files for all processors in the SAMD5x/E5x family, but is present in the Reference Manual. All possible TC mode configurations (`COUNT8`, `COUNT16`, `COUNT32`) in the associated PACs are missing this field.

jacobrosenthal, 2020-01-25:
We could use this issue as an opportunity to discuss how wed like to implement patching. In em32 I stole all the stm32 patching infrastructure. Its all based on python. That commit is here so you can get an idea of what it took https://github.com/em32-rs/efm32hg-pac/commit/97bc1eb248cf00d90b154e074d3e693976bd344c

sajattack, 2020-01-25:
Yeah, we've already slightly altered the SVDs from their originals. Would you be able to diff them with originals and turn those into patches?

jessebraham, 2020-01-25:
The patching from stm32-rs is in the process of being split out to [stm32-rs/svdtools](https://github.com/stm32-rs/svdtools)

proman21, 2020-01-25:
Not to be a issue purist but it would be better to file a separate issue for the SVD patching. Just cause there are some other things to chat about this issue.

Which is that the reference manual is not consistent w.r.t the DMA Oneshot functionality of the TC and TCC peripherals. Chiefly, because the TC peripheral does not have a DMA One-shot command in the Control B register but the relevant configuration option exists in the Control A register. So either the DMA One-shot functionality is supposed to match the TCC peripheral's implementation, or it is supposed to have no DMA One-shot functionality.

I'd recommend holding off on any changes to this part of the SVD until I can verify the actual behaviour _in silico_.

sajattack, 2020-07-29:
Is this fixed with the new svds?

PR #225 (2020-08-07): Add missing register field TCn.COUNTn.CTRLA.DMAOS for D5x/E5x devices
Closes #147.

(I had to learn XSL to do this; I think it's correct but if somebody could verify that'd be appreciated)
closed 2020-08-07 by sajattack