atsamd-rs/atsamd #164: Flaky behavior when using display on pygamer

johnbatty, 2020-02-01:
I'm seeing general flakiness when using the display on pygamer.
All the other examples seem to work fine.

Issues I have seen when tweaking the ferris_img demo:
- If I try to add a delay after drawing ferris, I get no display output
- If I change the background color when clearing the screen to something other than black or blue (a small number), I get no display output
- The board sometimes gets into a "bad state" where the ferris_img demo does not run.  If I flash another example, then go back and flash the same ferris_img demo again then that sometimes fixes it.

Has anyone else seen these or similar issues?
Does anyone have tips/guidance on how to set up low-level debugging for pygamer?  I'm new to this.  Is it possible to use OpenOCD with VSCode/Cortex-Debug extension?

jacobrosenthal, 2020-02-01:
I personally havent seen these, but havent been looking for them. Also as I said in the other thread, Im on a .6 alpha branch right now, Maybe check that out too?
https://github.com/jacobrosenthal/atsamd/tree/embedded-graphics-0.6

Really probably the issue will be in the embedded-graphics or display crate most likely.

Pygamer has debug headers, but youll have to solder them. You can find them on the pygamer eagle files adafruit provides. 

I use jlink for gdb
`JLinkGDBServer -if SWD -device ATSAMD51J19a -nogui`

You can panic and debug with jlinkrtt something like this
https://github.com/jacobrosenthal/atsamd/commit/b9b99d6e2c4f9a3f45dc0c7f0befe3b67f4b77e9

And get the output with `telnet localhost 19021`

And I THINK I was using this gdbinit
```
# print demangled symbols by default
set print asm-demangle on

# JLink
target extended-remote :2331
monitor flash breakpoints 1

monitor reset
load
```
And I think I was just adjusting the memory.x to blow away the bootloader as it complicates things to try to break past it.

johnbatty, 2020-02-01:
Thanks very much for the info - appreciated.  I'll see if I can get something set up.

johnbatty, 2020-02-19:
I've now got a JLink debug probe working with the PyGamer.  I don't see the flakiness any more, so I'm putting my issues down to unreliable flashing by the boot loader.

It took me a while to find settings that worked, so I think it might be useful for others if I checked in my `.gdbinit` and `openocd.cfg` files (I got it working with both OpenOCD and JLinkServer).  Would you be happy to accept a PR for those?

I'll close this issue.

jacobrosenthal, 2020-02-20:
Lets at least post in markdown here so we can see what you found to work
(or didnt work)

jessebraham, 2020-08-09:
@johnbatty said they would close this issue back in February and never did, so I'm going to just assume that was an oversight and close it now. If the issue persists please feel free to re-open this issue or open a new one!

closed 2020-08-09 by jessebraham