atsamd-rs/atsamd #169: Arduino MKR Wifi 1010 support

joonazan, 2020-02-06:
If I've understood correctly, the 1010 would work with the mkrzero code. But I need to use bluetooth, which works differently than on any other board on the 1010. How would I port that to Rust?

keithduncan, 2020-02-06:
The Arduino WiFi 1010 board has a u-blox NINA-W10 chip connected over SPI and can be used for 2.4GHz Bluetooth LE and WiFi.

The Arduino library for WiFi is https://github.com/arduino-libraries/WiFiNINA and for BluetoothLE is https://github.com/arduino-libraries/ArduinoBLE. I’m not aware of any Rust drivers for it yet, though it is something on my backlog 😅.

The firmware running on the NINA-W10 (ESP32) also open source and available at https://github.com/arduino/nina-fw. That’s the firmware it comes with out the box that you’d need to write a driver for. You could also fork that firmware or use a different ESP32 firmware and write a driver for _that_ instead if you wanted.

When I last looked at this I was able to [flash the NINA-W102 (ESP32) with the esp-at firmware](https://github.com/espressif/esp-at/issues/201#issuecomment-493847586) so that I could send AT commands over UART instead of binary commands over SPI. I haven’t picked this up in a while as I’ve been working with the u-blox SARA R410M LTE modem instead. Good luck!

joonazan, 2020-02-06:
I have tried ArduinoBLE, but it has the drawback that it waits for the
connection event synchronously. I also have to do I2C communication, so
currently the best workaround seems to be to write interrupt-driven I2C
code.

But my ideal solution would be to change the NINA firmware so that it
handles all the details of bluetooth and I only need to send characteristic
updates to it. I am not very familiar with embedded development. Would you
say that idea makes sense?

joonazan, 2020-02-06:
https://github.com/arduino/nina-fw/blob/3ba538f27ed56f4802911bb2b7d32e4f6b492e81/main/sketch.ino.cpp#L115

The firmware just makes bluetooth controllable via UART and then sleeps forever. Pretty wasteful.

I'll write my own firmware with the esp-idf NimBLE.

I doubt it is going to be generic enough to be worth including here, though. It will support writing to characteristics supporting read and notify and receiving updates on one writable characteristic.

joonazan, 2020-03-13:
@keithduncan I now have a working Arduino library. https://github.com/joonazan/nina-fast-bluetooth

A Rust version of the SAMD side would be pretty straightforward to make.

I also noticed that the MKR WiFi is a very awkward board. NINA is harder to flash but because of its specs it should be the main computer. It has 520kB of RAM and a dual core processor that is at least ten times faster than the SAMD. It has stack traces and FreeRTOS if using esp-idf.

I think it would be ideal to make a build script that makes it easy to flash Rust onto the NINA and SAMD in one go.

keithduncan, 2020-03-14:
> a dual core processor that is at least ten times faster than the SAMD

It’s funny isn’t it :smile:

The NINA W10 has an Xtensa core which has preliminary [rust support](https://github.com/MabezDev/rust-xtensa).

Nice work getting an esp32 app running 👍

sunsided, 2025-01-01:
See also #606

ianrrees, 2025-03-11:
Closing this as there's now a PR

closed 2025-03-11 by ianrrees