atsamd-rs/atsamd #175: GCLK observe on port pin

sameer, 2020-03-29:
For the MKR Vidor 4000 (samd21g18a), gclk0 needs to be observed on a port pin (PA27) so that the 48MHZ clock can be sent to the onboard FPGA.

With the current clock configuration functions, there is no way to set the output enable/output off values which are used to do this.

I made a feature to enable gclk0 output on my end for now:  https://github.com/sameer/atsamd/commit/7742311cbc3bcf75bd756ad0c9e722ed2d877b43 

I want to make a more generic pull request to allow observing any GCLK. I could add these extra parameters to the pub fn's and document them, but that'll break the current interface.

If you have any suggestions on how this should be implemented, I would really appreciate them.

sajattack, 2020-03-29:
I added this when I made samd51 but I guess I never backported to samd21. https://github.com/atsamd-rs/atsamd/blob/master/boards/metro_m4/examples/clock_out.rs

sameer, 2020-03-29:
> I added this when I made samd51 but I guess I never backported to samd21.
> 
> https://github.com/atsamd-rs/atsamd/blob/master/boards/metro_m4/examples/clock_out.rs
> 
> https://github.com/atsamd-rs/atsamd/blob/513f24d9cb3a832bc85b3566afa7709684cb07b4/hal/src/samd51/clock.rs#L117
> 

Cool, let me know if you remember any other stuff that needs to be backported and I'll see what I can do.

> It doesn't really need an extra parameter in the function I don't think because it only turns on if the pin is in the right output mode.

Oh ok, that makes things a lot simpler then! I'll go ahead and make a PR for this.

PR #176 (2020-03-29): GCLK output enable for samd21, samd11
Closes #175.
closed 2020-03-29 by sajattack