atsamd-rs/atsamd #182: SAMD21 NVMCTRL.CTRLB.MANW = 1

sameer, 2020-04-02:
In [errata 1.11.2](http://ww1.microchip.com/downloads/en/DeviceDoc/SAM_D21_%20DA1_Family_Silicon_Errata_DS80000760E.pdf#_OPENTOPIC_TOC_PROCESSING_d137e8058), there's some discussion of how all the chips have MANW set to 0 but it should be 1 by default. MANW turns off automatic writes to NVM.

If a pointer that's supposed to write to the data region addresses the NVM region and writes to it, it would actually change the NVM automatically. This isn't as big of a concern for Rust I think, but we should still address it.

[Adafruit](https://github.com/adafruit/uf2-samdx1/blob/master/src/startup_samd21.c#L264) and [Arduino](https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/startup.c#L300) have both corrected this and I'm wondering where the best place to put this might be for atsamd. It could be added during clock configuration where the crate has access to NVMCTRL, like [another project does for samc21](https://github.com/ataradov/mcu-starter-projects/blob/125ba75f592833ed72df71cf41b2c6ce6b47350b/samc21/main.c#L132).

sajattack, 2020-04-02:
Yeah clock configuration is probably a decent spot because it's always run by user programs.

sameer, 2020-04-02:
> Yeah clock configuration is probably a decent spot because it's always run by user programs.

Cool I'll PR that.

PR #183 (2020-04-02): NVMCTRL.ctrlb.manw = 1, fixes #182
Fixes #182
commit 868d9d19 (2020-04-02): NVMCTRL.ctrlb.manw = 1, fixes #182
closed 2020-04-02 by sajattack