atsamd-rs/atsamd #195: Issues blocking `update.sh` on master

gkelly, 2020-05-11:
I recently started adding support for ATSAMD51P19A. While I was doing this I ran `update.sh` and noticed that at least one other ATSAMD51 doesn't actually build at head. I started digging in to this and discovered a few issues.

- [ ] `svd2rust` 0.17.0 doesn't support dimable names (`CHANNEL[%s]`, for example). This is because in December of 2019 the `svd-parser` package was patched to enforce naming rules, but it uses the pre-CMSIS 1.3.3 naming rules. I opened https://github.com/rust-embedded/svd/pull/119 to fix this.
- [ ] `svd2rust` is missing adding some `use` declarations in some modules. Haven't dug into this yet.
- [ ] Some register reset values generated by `svd2rust` are bigger than their containing types. This doesn't get caught until after `svd patch`, `svd2rust`, and `form` are run. I opened https://github.com/rust-embedded/svd/pull/120 to catch these in `svd-parser`. This doesn't _block_ getting `update.sh` running again, but it's a quality-of-life improvement that would help catch breakages early.

jessebraham, 2020-05-11:
Thanks for opening this issue. I am currently working on at least some related problems, just haven't had a ton of time for this project lately unfortunately. Probably should have opened an issue myself, so thanks for taking care of that.

I have a few patches to be applied to the SVDs locally, however I need to patch `svdtools` one more time to allow modifying nested clusters; once there is support for that, the SVDs can be fully patched (as far as I know) and that will help at least some issues.

Regarding the clusters, AFAIK you are correct in that `svd2rust` has some issues processing them currently, so we're sort of blocked on that too.

I will try to list out what has been done and what needs to be done (other than what you've posted) later tonight or this week when I get a bit more time.

gkelly, 2020-05-11:
> Thanks for opening this issue. I am currently working on at least some related problems, just haven't had a ton of time for this project lately unfortunately. Probably should have opened an issue [...]

I know that `svd patch` has a hard time touching the descriptions of clusters, but `svd-parser` at tip can parse empty descriptions fine. I think getting `svd-parser` fixed to handle in-spec names (`CHANNEL[%s]` is valid in CMSIS 1.3.3+) means not having to modify clusters, doesn't it?

> Regarding the clusters, AFAIK you are correct in that `svd2rust` has some issues processing them currently, so we're sort of blocked on that too.
>
> I will try to list out what has been done and what [...]

gkelly, 2020-05-12:
I tracked down the cluster generation issue in `svd2rust` and have put up a PR that hopefully addresses it: https://github.com/rust-embedded/svd2rust/pull/444.

gkelly, 2020-05-12:
Those changes are merged. There are still SVD patches needed (I believe) to make things work with tip `svd2rust` and tip `svd-parser`. I don't know what the release cadence is for those tools, but hopefully it won't be too long before those changes are in a release and can be referenced from `update.sh`. I'll update the checklist to capture that.

gkelly, 2020-05-13:
The only error I haven't been able to `svd patch` out with a tip `svd2rust` and tip `svd-parser` is `ATSAMD11C14A`'s:

```
[ERROR svd2rust] In device `ATSAMD11C14A`                                                                                                                                                                                                                                                   
                                                                                                                                                                                                                                                                                            
    Caused by:                                                                                                                                
        0: In peripheral `USB`                                                                                                                
        1: In cluster `DEVICE`                                                                                                                
        2: In register `FSMSTATUS`                                                                                                            
        3: Parsing field #0                                                                                                                   
        4: In field `FSMSTATE`                                                                                                                
        5: Value 64 out of range 0..64
```

It's in the `DEVICE` cluster, which I'm not sure if `svdtools` can currently get inside very well.

wez, 2020-05-13:
> I don't know what the release cadence is for those tools, but hopefully it won't be too long before those changes are in a release and can be referenced from `update.sh`. I'll update the checklist to capture that.

Hopefully they're better these days, but in the past it wasn't great; I had a PR that took a year to get merged and shipped, so I used to maintain a fork for atsamd.  I'd suggest pointing to your fork (or we can make a fork under the `atsamd-rs` org if you don't want to own that repo) if you feel like it is taking a while and getting in the way of the stuff you really want to work on; you can see how we used to do this in the history of `update.sh`.

gkelly, 2020-05-14:
> Hopefully they're better these days, but in the past it wasn't great; I had a PR that took a year to get merged and shipped, so I used to maintain a fork for atsamd. I'd suggest pointing to your [...]

Seems reasonable to me, thanks for the advice. With those two patches I just put up things are very close to building. The last broken thing I haven't figured out yet is `ATSAMD11C14A`, where the SVD error is inside the `USB.DEVICE` cluster. That might take an `svdtools` patch to fix. @jessebraham do you have any tricks up your sleeve on that front, or were you looking at a different part?

jessebraham, 2020-05-16:
Sorry for the delay, I've had a very busy week. Thank you for putting the time and effort into this, I do appreciate it!

I'm glad to hear you got a PR merged in `svd2rust`. I know the release cycle on some of the **rust-embedded** crates can be quite lengthy, so hopefully we don't need to wait too long for your changes to be released. I do think that (at least temporarily) maintaining our own fork probably isn't the worst idea, as we could potentially be waiting quite some time and it's reasonably easy to revert back to the upstream version at a later date. I'll leave that decision up to you/others though :)

You are correct that `svdtools` does not handle clusters especially well in its current state. I did get [a PR merged](https://github.com/stm32-rs/svdtools/pull/27) which enabled modifying clusters, but it has yet to released. It might help, might not. I think there are also some nested clusters in our SVDs which cause further problems, and I don't know how well (or if at all) they're handled by `svdtools` unfortunately.

If I missed anything or you need anything else just let me know, I should be mostly available throughout this weekend.

gkelly, 2020-05-16:
> Sorry for the delay, I've had a very busy week. Thank you for putting the time and effort into this, I do appreciate it!

No worries, thanks for taking the time to review and respond!

> I'm glad to hear you got a PR merged in `svd2rust`. I know the release cycle on some of the **rust-embedded** crates can be quite lengthy, so hopefully we don't need to wait too long for your [...]

I've updated https://github.com/atsamd-rs/atsamd/pull/197 to use a pinned fork of `svd2rust`.

> You are correct that `svdtools` does not handle clusters especially well in its current state. I did get [a PR merged](https://github.com/stm32-rs/svdtools/pull/27) which enabled modifying clusters, [...]

Using the https://github.com/gkelly/svd2rust/commit/2bbb60590096bcb67c91f38bedd1f63f98132abe version of `svd2rust` and having https://github.com/atsamd-rs/atsamd/pull/196 landed I believe only `ATSAMD11C14A` fails to generate, and it's because of the wrong register size inside of the `USB.DEVICE` cluster.  I'll take a look to see if your merged `svdtools` change can handle that modification.
 
> If I missed anything or you need anything else just let me know, I should be mostly available throughout this weekend.

Thanks, I appreciate the reviews and responses!

gkelly, 2020-05-17:
I've spent some time reading the `svdtools` source. It has some opinions about the structure of SVDs that are incorrect, such as: clusters are always contained by peripherals, clusters are never contained by clusters, things can't be named starting with an underscore. It also doesn't currently actually support modifying registers in clusters.

In the `ATSAMD11C14A` case the invalid value is contained in a register inside a cluster inside a device. It's a single-character change, but I think it's going to take a bit of hacking on `svdtools` to resolve.

I almost wonder if `svdtools` is taking the wrong approach in general. `svdtools` processes XML to XML. I wonder if `xsltproc` isn't a better way to describe the transforms. I don't think there are any transforms `xsltproc` _can't_ perform.

gkelly, 2020-05-19:
I wrote #199 to get a feel for how (un)ergonomic writing XSLT is instead of `svd patch` yamls. Not that bad, it turns out. With #199 `update.sh` can run at master. I have built all of the generated `pac` crates as well and they all build cleanly.

blm768, 2020-07-20:
Did [this commit](https://github.com/atsamd-rs/atsamd/commit/0e784c99d201cd955abeb04d739ad7e83f120726) fix the issues with ATSAMD11C14A devices, or is there still more work to be done on them?

jessebraham, 2020-07-21:
The ATSAMD11C14A seems to indeed have been patched [here](https://github.com/atsamd-rs/atsamd/blob/master/svd/devices/atsamd11c14a.xsl#L5-L7), unless I'm misunderstanding or this is for something else.

I was able to check out the repository, patch the SVDs, build the PACs and HAL, and build the example crates for the `samd11_bare` board. Is there anything left to do other than generate/release new PACs?

sajattack, 2020-07-21:
We need to rewrite the HAL to support the new PACs

closed 2020-07-29 by sajattack