atsamd-rs/atsamd #203: Type erasure for GPIO

AdminXVII, 2020-06-09:
As per https://rust-embedded.github.io/book/design-patterns/hal/gpio.html#pin-types-provide-methods-to-erase-pin-and-port-c-erased-pin, HAL crates should have means to downgrade the pins to a type-erased version. This is particularly useful for storing a list of pins in an array rather than a tuple.

I didn't find a way to perform this, so I'd like to request for help and/or for this feature. Thanks for your time.

jaredwolff, 2020-06-30:
Here's a link to a similar discussion. Looks like it's already implemented for some STM32 HAL: https://users.rust-lang.org/t/how-can-i-create-an-array-of-pins/44514

rubberduck203, 2020-08-03:
`stm32f3xx-hal` implements it like this. 

https://github.com/stm32-rs/stm32f3xx-hal/blob/master/src/gpio.rs#L140

https://github.com/stm32-rs/stm32f3xx-hal/blob/master/src/gpio.rs#L396

jessebraham, 2020-08-03:
Thanks for opening this issue.

I would also like to see type erasure for GPIO, as I have a handful of projects which could benefit from it myself. If somebody wants to take a swing at it that would be awesome! If not, I can take a look at implementing this feature whenever I get a chance; unfortunately I'm fairly busy with some other projects currently.

blm768, 2020-09-13:
It seems like the proposed changes for #214 would likely help with this; a common `Pin` trait seems like it would make a generic implementation of type erasure easier.

bradleyharden, 2020-10-26:
@Sympatron,

I agree with @blm768 that the changes in #214 make this easier. I didn't have any need for type-erasure myself, so I hadn't already implemented it. But since you're working on it, I thought I would demonstrate.

Here is the latest [version](https://github.com/bradleyharden/atsamd/blob/gpio_refactor/hal/src/common/thumbv7em/gpio_new.rs), with the type-erased pin found [here](https://github.com/bradleyharden/atsamd/blob/9c3cfd8dd96609ebda3fcdc15c9c9ed1bb46524f/hal/src/common/thumbv7em/gpio_new.rs#L783). I was able to easily duplicate the entire pin API for type-erased pins. You can also convert between type-level pins and value-level pins, see [here](https://github.com/bradleyharden/atsamd/blob/9c3cfd8dd96609ebda3fcdc15c9c9ed1bb46524f/hal/src/common/thumbv7em/gpio_new.rs#L1005).

So far, @jessebraham and @sajattack have been resistant to accept the proposed changes in #214 until we can lump them in with other breaking changes in the move to 1.0. But I would counter that continuing to build on top of the current code will only get more difficult, and it will mean we have to re-implement more pieces later. Moreover, the macros have shown themselves to be a barrier to entry for contributors, so we might attract more contributions with an easier-to-understand code base.

sajattack, 2020-10-26:
We haven't been resistant to accept the changes, you haven't even PR'ed them yet.

bradleyharden, 2020-10-26:
@sajattack, sorry, I probably should have phrased that differently. I was specifically talking about some conversations with @jessebraham, probably on Matrix. He seemed to prefer waiting for 1.0, but maybe you have a different opinion? Or maybe I misunderstood?

I think my replacement `gpio`, `pads` and `spi` modules are all just about ready for a PR. I've been using them on a project with good results so far. But integration with the rest of the code is a much different story. I'm not really sure how to approach it.

At first, I assumed you would want to completely swap the old modules for the new modules. But that would be severely breaking and leave everything in limbo until it could all be updated. But I realized just now that both the old and new modules could live side by side for the foreseeable future. That would allow a smoother migration, but it would be piecemeal, and it might be confusing for users to have two different modules for the same purpose.

If you're willing to accept PRs, then I'll start making them. Just let me know how you want to approach integration.

Sympatron, 2020-10-26:
@bradleyharden, I was aware of the fact that the gpio module might be reworked, but I thought it might be useful until that happens. Honestly it wasn't that much work and I used it as an exercise to get to know the hal.
I should probably read #214, but it's length was kind of daunting tbh...

If you plan to use them side by side this should be very clearly documented. Nothing is more confusing to beginners than multiple APIs for them same thing. That always make you think you don't really understand what's going on.

Maybe the old API can be marked as deprecated and at 1.0 it can be removed or something. That way people can experiment with new API before 1.0.

bradleyharden, 2020-10-26:
@Sympatron, yeah, #214 did get quite long. @blm768 gave some good feedback. I appreciate that he was willing to go back and forth with me.

In general, I never know how I should approach something like that. Is it better to be more verbose and walk through the code and motivation behind key decisions? Or should I just throw up a link to the code and let people go right to the source? Or maybe something in between? I'm not sure what option is the most digestible. Let me know if you have any suggestions.

If we opt for the side-by-side approach, I can work to make the documentation clear.

bradleyharden, 2020-11-19:
#302 has been merged and contains a type-erased API. Check out the [dynpin](https://github.com/atsamd-rs/atsamd/blob/master/hal/src/common/gpio/v2/dynpin.rs) module.

AdminXVII, 2020-11-19:
Thanks, that looks great! Closing this issue as it seems to be dealt with.

closed 2020-11-19 by AdminXVII