atsamd-rs/atsamd #213: Instantiating sercom with frequency is misleading at best and broken at worst

BroderickCarlin, 2020-07-20:
After spending some time attempting to debug a SPI issue I came to the realization that the instantiated SPI bus's frequency was not inline with what had been specified during instantiation. Experimenting with passing in different `freq` values during initialization resulted in non-trivial changes to the realized SPI bus's frequency. Digging into [the code](https://github.com/atsamd-rs/atsamd/blob/48b6624c0ea51d9ca568d5b206165ece45a31c6a/hal/src/samd21/sercom/spi.rs#L136) it became evident that the frequency parameter passed in was being converted to a baud rate internally. This hidden conversion does not appear to be doing proper bound checking and as such seemingly valid inputs result in non-discriminate outputs. 

For example, let's imagine a user wanted to have an exceptionally slow clock for some sort of visualization of SPI traffic. The code as written today would entirely allow for a frequency of `1.hz()` to be passed in. However, if we work through the math of how this value would be interpreted we can see that it would not actually yield a bus frequency of, or near, 1hz. 

My suggestion would be to revisit this math and add proper bounds checks

sajattack, 2020-07-20:
Sounds like a good idea.

sajattack, 2020-07-20:
@TDHolmes since you have the most recent experience mucking with the timers, would you be willing to take this on? I'm thinking making TimerParams fallible (return a Result) is probably a good approach.

BroderickCarlin, 2020-07-20:
As a sort of later thought on this, it may be beneficial to have an accessor method to see what the actual frequency is that gets set. Adding bounds checks, or making it fallible, in conjunction with allowing a developer to access what the actual clock speed is that gets set would be a huge improvement imho. 

@TDHolmes, thoughts?

sajattack, 2020-07-24:
Haha I just realized you were talking about sercoms not timers.

TDHolmes, 2020-07-25:
haha, I mean it is somewhat related :p The proposed solution makes sense. I'm down to take a crack at it

TDHolmes, 2020-07-25:
Looks like we could use [TryFrom](https://doc.rust-lang.org/beta/core/convert/trait.TryFrom.html) to attempt the conversion, and error out is it won't fit. Do we have a MSRV? Looks like this was added in 1.34

sajattack, 2020-07-31:
I'm actually not sure if an error is the best fit. Maybe just let the user know what they're actually getting.

BroderickCarlin, 2020-07-31:
yeah; personally I think just letting the user know what frequency was actually set would be very beneficial

TDHolmes, 2020-08-02:
seems clunky to me... you start a timer and then always have to check it after? If we know we've (kinda) failed, why not tell the user right away?

Either way, an initial PR just to get the frequency back seems reasonable.

sajattack, 2020-08-03:
Sometimes a closest fit is better than a panic.

TDHolmes, 2020-08-03:
I agree, but it doesn't have to be a fatal error. I was thinking something along the lines of:
```rust
pub enum SpiError {
    BaudrateTruncated,
};

pub fn new(...) -> (SpiMasterN, Option<SpiError>) { ... }
```

However, now that I write it out, it feels not very clean, so I think I'm in agreement with you guys now. Something like `Result` isn't really built for non-fatal errors, and we don't need to modify an API.

sajattack, 2020-08-07:
@BroderickCarlin can this be closed now?

BroderickCarlin, 2020-08-09:
Solved by: https://github.com/atsamd-rs/atsamd/pull/224

closed 2020-08-09 by BroderickCarlin