atsamd-rs/atsamd #215: Would This work with ADAFRUIT FEATHER M0 WIFI - ATSAMD21 + ATWINC1500?

bobhenkel, 2020-07-24:
I'm really new to embedded so keep that in mind. 

Would this be able to make use of this board https://www.adafruit.com/product/3010 and make use of the wifi?

Thanks!

sajattack, 2020-07-24:
I'm not aware of a rust crate for the ATWINC1500.

sajattack, 2020-07-24:
@fionawhim is working on wifi support for boards like the pyportal that use esp32 for wifi https://github.com/fionawhim/wifinina

fionawhim, 2020-07-24:
Yep. Unfortunately the ESP32 WiFiNINA looks to share nothing with the ATWINC1500 besides both being communicated with over SPI. (From the look of it, Adafruit indicates that driving a ATWINC1500 is so complicated that they can’t fit it in Python on an M0?)

My stuff will work with PyPortal and, presumably, anything that Adafruit markets as "AirLift."

closed 2020-07-24 by fionawhim

bobhenkel, 2020-07-24:
Thanks for the quick response @sajattack and @fionawhim