atsamd-rs/atsamd #247: RFC: create variant features for configuration common to a physical package

jessebraham, 2020-09-03:
In #243 @bradleyharden had suggested that we create variant features (eg. for `samd51g`, `samd51j`, `samd51n` and `samd51p`) to cover the different pinouts. I don't think this is a bad idea personally, and doing so could potentially clean up our `#[cfg()]`s quite a bit, especially as we continue to add support for new devices.

Does this seem like a worthwhile venture? Are there alternative approaches we could take? Thoughts?

blm768, 2020-09-13:
Another option might be to add features for specific hardware units that are present, i.e. a `sercom7` feature for all devices that have at least eight SERCOMs. Not sure if that's really better, though.

Sympatron, 2020-10-23:
#283 implements this. I think this can be closed now.

closed 2020-10-23 by sajattack