atsamd-rs/atsamd #250: HAL for a specific board - cargo.toml - how to?

nebelgrau77, 2020-09-11:
Not an issue, but a noobish question: I have some experience with embedded Rust on STM32, and I want to experiment with SAMD, too. I do sometimes struggle with Rust's module system, though. 

Here's my problem: I can build all the examples in bulk all right, but I only have an Arduino 33 IoT and that's what I want to focus on. The examples for my board specify the HAL in Cargo.toml as being retrieved from a local path, and I assume it's the lib.rs within arduino_nano33iot folder, that is later called in the code as

```rust
extern crate arduino_nano33iot as hal
```
Here's my question: how can I specify this crate in my Cargo.toml without saving the lib.rs locally? I just want the compiler to get it automatically. On crates.io I can find this: atsamd21g18a = "0.7.0" - is the arduino_nano33iot basically a user friendly configuration on top of the main crate? 

Thanks in advance for some help/clarification, I'm sure it will help other users who want to build code for specific boards.

sajattack, 2020-09-11:
https://crates.io/crates/arduino_nano33iot
You can use this or do your own with the samd21g18a crate. I don't see a reason not to use the board support though.

nebelgrau77, 2020-09-11:
Great, thanks! It worked perfectly.

closed 2020-09-11 by nebelgrau77