atsamd-rs/atsamd #251: SSD1306 + Arduino 33 IoT

nebelgrau77, 2020-09-12:
I'm trying to get the OLED to work with Arduino 33 IoT. The only examples I found were for ItsyBitsyM0 and FeatherM0. They initialize I2C with "i2c_master" function, which is not defined in the Arduino 33 IoT crate. That's not a problem, but I don't think I got it right: https://github.com/nebelgrau77/Arduino33IoT_OLED_test
It seems that the display can't get initialized, because commenting the disp.init() line makes the LED blink, i.e. it must be halting at that point. There's no I2C activity on the pins, as checked with a logic analyzer. Is it the clock setting, or the wrong SERCOM?

Thanks!

sajattack, 2020-09-12:
B8 and B9 appear to be on SERCOM4. You can see here: http://ww1.microchip.com/downloads/en/DeviceDoc/SAM_D21_DA1_Family_DataSheet_DS40001882F.pdf#_OPENTOPIC_TOC_PROCESSING_d138e23050 I'd appreciate if you contributed the i2c_master function back to the BSP once you get it working.

nebelgrau77, 2020-09-12:
OK, so that would explain it, and I understand that the necessary code modification would be this:

```rust
   let gclk0 = clocks.gclk0();
    let i2c = hal::sercom::I2CMaster4::new(
        &clocks.sercom4_core(&gclk0).unwrap(),
        KiloHertz(100),  // I lowered the frequency just to be on the safe side here
        peripherals.SERCOM4,
        &mut peripherals.PM,
        pins.sda,
        pins.scl,
        
    );
```
It's still not working, though :/

sajattack, 2020-09-12:
I think you need `.into_pad(&port)`
https://github.com/atsamd-rs/atsamd/blob/0f633d615282c2ec9f8932e45167d2069f6e7ff7/boards/feather_m0/src/lib.rs#L138

nebelgrau77, 2020-09-12:
Got it to work, basically by adding the i2c_master to my local fork of arduino_nano33iot! It's working, now I only need to figure out how to contribute :)

sajattack, 2020-09-12:
Awesome! :D

closed 2020-09-12 by sajattack