atsamd-rs/atsamd #254: Add uart, i2c_master, spi_master and usb_allocator for all BSPs

sajattack, 2020-09-12:
Should be fairly simple to do, just time consuming, and would have prevented issues like #251

jessebraham, 2020-09-13:
If you spot any errors in the table below, or any changes have been made, please post in this issue and I will update the table.

| Board | `i2c_master` | `spi_master` | `usb_allocator` |
|:------|:-------------|:-------------|:----------------|
| arduino_mkrvidor4000 | ❌ | ❌ | ❌ |
| arduino_mkrzero | ❌ | ❌ | ✔ |
| arduino_nano33iot | ✔ | ✔ | ✔ |
| circuit_playground_express | ✔ | ❌ | ❌ |
| edgebadge | ❌ | ❌ | ❌ |
| feather_m0 | ✔ | ✔ | ✔ |
| feather_m4 | ✔ | ✔ | ✔ |
| gemma_m0 | ❌ | ❌ | ❌ |
| itsybitsy_m0 | ✔ | ✔ | ✔ |
| itsybitsy_m4 | ✔ | ✔ | ❌ |
| metro_m0 | ✔ | ✔ | ❌ |
| metro_m4 | ✔ | ✔ | ✔ |
| pfza_proto1 | ✔ | ✔ | ❌ |
| pygamer | ❌ | ❌ | ❌ |
| pyportal | ✔ | ✔ | ❌ |
| samd11_bare | ✔ | ❌ | ❌ |
| samd21_mini | ❌ | ❌ | ❌ |
| serpente | ❌ | ❌ | ❌ |
| sodaq_one | ✔ | ✔ | ❌ |
| sodaq_sara_aff | ❌ | ❌ |❌  |
| trellis_m4 | ✔ | ❌ | ❌ |
| trinket_m0 | ✔ | ❌ | ✔ |
| xiao_m0 | ✔ | ✔ | ✔ |

sajattack, 2020-09-13:
You missed uart

jessebraham, 2020-09-13:
Yes I just noticed that, thanks. Will update when I've gone through all the BSPs (again).

Sympatron, 2020-10-22:
Are you talking about the actual BSP implementation (i.e. convenience functions to initialize I2C, etc.) or are you talking about examples?

jessebraham, 2020-10-22:
This is referring to the convenience functions, you can see [an example here](https://github.com/atsamd-rs/atsamd/blob/master/boards/trinket_m0/src/lib.rs#L188-L212).

jessebraham, 2020-11-24:
Many of our BSPs have opted to use pin `Sets`, and only some include the tracked helper functions.

Should we consider the `Sets` approach to be adequate for being marked completed here? [wio-terminal](https://github.com/atsamd-rs/atsamd/tree/master/boards/wio_terminal) is an example of a BSP where it doesn't necessarily make sense to include the above functions, IMO.

jacobrosenthal, 2020-11-27:
yeah for bsps with Sets model solves the same problem better.

You could argue for both, but I actually killed these functions off last year because I felt like either I was duplicating code, or the one helper fn was calling the other making the bsp even more inscrutable to newcomers.

In sets you even get access to functionality equivalent `i2c_master` assuming for some reason you're not just using the Set you were given from `.split` by constructing the Set and `init` it

```
let i2c = I2C {
    sda,scl
};
i2c.init()l
```

sajattack, 2020-11-27:
Yeah I would consider either acceptable and equivalent.

quentinmit, 2021-04-22:
I have started working on implementing a new BSP (#431) and running into difficulty structuring the convenience functions for it. For boards with well-defined peripherals, the `Sets` mechanism makes sense - you can split off the I2C pins into a separate struct that represents I2C. But for boards with multi-use pins such as this one, where you might want to use them as UART pins or you might want to use them as GPIOs, I don't think `Sets` helps. Also, pin aliases don't seem to help much because you still have to know the base name of each pin - so e.g. my BSP needs `let ss: hal::BaseSlaveSelect = pins.d18.into();` if you want to use the base slave select pin, which means that the user has to know that pin member is called d18.

Finally, the other problem I've run into is that it doesn't seem possible to take a pin and turn it into a `Sercom`/pad in `gpio::v2` - I had to hardcode `PB23` and `PB22` as types in my `uart` function even though the `bsp_pins!` macro created `UartRx` and `UartTx` types.

I think the ideal interface would be to somehow tie the convenience functions directly to the `Pins` struct, so splitting them off and constructing the `UART`/`UsbBusAllocator`/etc. happens in a single method call.

TDHolmes, 2021-10-16:
updated status @jessebraham, and an attached script to automatically check

|Board|`i2c_master`|`spi_master`|`usb_allocator`|`uart`|
|-|-|-|-|-|
|~feather_m0~|✔|✔|✔|✔|
|~metro_m4~|✔|✔|✔|✔|
|wio_terminal|❌|❌|❌|❌|
|edgebadge|❌|❌|❌|❌|
|~trinket_m0~|✔|✔|✔|✔|
|arduino_mkrvidor4000|❌|❌|✔|❌|
|neo_trinkey|❌|❌|✔|❌|
|itsybitsy_m4|✔|✔|❌|✔|
|pygamer|❌|❌|❌|❌|
|sodaq_one|✔|✔|✔|❌|
|~wio_lite_mg126~|✔|✔|✔|✔|
|gemma_m0|❌|❌|❌|✔|
|atsame54_xpro|❌|❌|❌|❌|
|~feather_m4~|✔|✔|✔|✔|
|~metro_m0~|✔|✔|✔|✔|
|circuit_playground_express|✔|✔|❌|❌|
|grand_central_m4|❌|❌|❌|❌|
|serpente|❌|❌|❌|✔|
|arduino_mkr1000|❌|❌|✔|✔|
|arduino_mkrzero|❌|❌|✔|❌|
|samd11_bare|✔|❌|❌|✔|
|sodaq_sara_aff|✔|❌|✔|✔|
|samd21_mini|❌|❌|❌|❌|
|~arduino_nano33iot~|✔|✔|✔|✔|
|~itsybitsy_m0~|✔|✔|✔|✔|
|~xiao_m0~|✔|✔|✔|✔|
|trellis_m4|✔|❌|❌|✔|
|~qt_py_m0~|✔|✔|✔|✔|
|pyportal|✔|✔|❌|✔|
|~p1am_100~|✔|✔|✔|✔|
|pfza_proto1|✔|✔|❌|❌|
-----------

TDHolmes, 2022-04-01:
`pygamer` and `wio_terminal` actually also have these functions, they're just wrapped up in pin sets. That means all tier 1 boards have these functions. tier 2 boards are more best-effort, so I don't see much benefit in tracking this anymore. If folks disagree, we can re-open

closed 2022-04-01 by TDHolmes