atsamd-rs/atsamd #268: ADC problem: unresolved import (Arduino Nano 33 IoT)

nebelgrau77, 2020-09-17:
I'm trying to port some useful examples from other boards to Arduino 33 IoT, and I ran into a problem I don't really understand.

In the adc.rs example for FeatherM0 there is the following import:

```rust
use hal::adc::Adc;
```

but this will not compile for the Arduino crate, despite the two crates being basically identical in terms of exposing the `atsamd_hal`, `samd21`, `common`, etc. Also the use of the `hal` is identical between the crates, the FeatherM0 example does it as follows:
```rust
extern crate feather_m0 as hal;
```
so mine would be:
```rust
extern crate arduino_nano33iot as hal;
```

The compiler error is:
```rust
error[E0432]: unresolved import `hal::adc`
 use hal::adc::Adc;
               ^^^ could not find `adc` in `hal`
 ```

What am I missing here?

jessebraham, 2020-09-17:
I believe compiling with `--features=unproven` should fix that issue. Once `embedded-hal` reaches `1.0.0` then the `unproven` feature will be removed.

nebelgrau77, 2020-09-17:
That was it, thank you! I guess it's my first time to compile with the 'features' flag :)

closed 2020-09-17 by nebelgrau77