atsamd-rs/atsamd #269: ADC type for RTIC resources - help needed

nebelgrau77, 2020-09-18:
I'm trying to learn a little bit of RTIC. So far managed to successfully add UART to the blinky example, and it works. Now I would also like to read ADC. I can't figure out the right type to define for the ADC resource. I tried the trick with assigning:
```rust
let _: u8 = adc;
```
and the compiler tells me that it's not an u8, but instead
```rust
note: expected type `u8`
            found struct `arduino_nano33iot::adc::Adc<arduino_nano33iot::atsamd21g18a::ADC>`
```
But that won't work. I guess it's the re-exports make it complicated. Here's my code: 

https://gist.github.com/nebelgrau77/6f158d3ee406925d864673a885023039

I'd be happy to hear some advice!

Thanks!

sajattack, 2020-09-18:
Does this work? `hal::adc::Adc<hal::pac::ADC>`

nebelgrau77, 2020-09-18:
It does compile, so I guess it's right, thanks! Still don't know how to make my code work exactly, but I will figure it out :)

closed 2020-09-18 by nebelgrau77