atsamd-rs/atsamd #270: the trait `hal::prelude::_embedded_hal_digital_InputPin` is not implemented

michaelfletchercgy, 2020-09-18:
I am attempting to use a hal InputPin my with feather_m0 but am getting an error.

```
let mut peripherals = Peripherals::take().unwrap();
let mut parts = peripherals.PORT.split();
let mut in_pin = parts.pa16.into_floating_input(&mut parts.port);

if let Ok(_is_high) = in_pin.is_high() {
    // do something
}
```

I don't understand how to resolve those traits.

```
error[E0599]: no method named `is_high` found for struct `Pa16<Input<Floating>>` in the current scope
   --> src/main.rs:21:34
    |
21  |       if let Ok(_is_high) = in_pin.is_high() {
    |                                    ^^^^^^^ method not found in `Pa16<Input<Floating>>`
    | 
   ::: /home/michaelfletcher/.cargo/registry/src/github.com-1ecc6299db9ec823/atsamd-hal-0.9.0/src/common/thumbv6m/gpio.rs:498:1
    |
498 | / port!([
499 | |     Pa0: (pa0, 0),
500 | |     Pa1: (pa1, 1),
501 | |     Pa2: (pa2, 2),
...   |
563 | |     Pb31: (pb31, 31),
564 | | ]);
    | |   -
    | |   |
    | |___doesn't satisfy `_: embedded_hal::digital::v2::InputPin`
    |     doesn't satisfy `_: hal::prelude::_embedded_hal_digital_InputPin`
    |
    = note: the method `is_high` exists but the following trait bounds were not satisfied:
            `Pa16<Input<Floating>>: hal::prelude::_embedded_hal_digital_InputPin`
            which is required by `Pa16<Input<Floating>>: embedded_hal::digital::v2::InputPin
```

I've tried to update to the latest of everything and create the smallest possible reproduction.  Any help or hints would be appreciated.  I have the error minimized in this repro: https://github.com/michaelfletchercgy/repro-compiler-error

sajattack, 2020-09-18:
I think you need the unproven feature, `--features=unproven`, this is going away soon though so it should be less confusing in the future.

michaelfletchercgy, 2020-09-18:
OHHH.  Unproven in the feather_m0 crate.  I already had unproven in the embedded-hal crate.

closed 2020-09-18 by michaelfletchercgy