atsamd-rs/atsamd #272: Questions about a proposed refactoring of the clock module

bradleyharden, 2020-09-28:
The existing implementation of the `clock` module is not very configurable, at least not for the `SAM51D` series. In the project I'm working on, I had to resort to setting up the clocks with C-style code. That works well enough for now, but I started trying to figure out how I would refactor the `clock` module to be more configurable.

I think I have a reasonable structure, but I ran into an issue that I think will cause some ergonomic headaches. I was wondering if someone else might have some insight.

First, I should probably give a brief overview of the `GCLK` peripheral, in case people aren't familiar. The `GCLK` peripheral has 12 clock generators. Each clock generator takes a source clock, optionally divides it, and then distributes it to any of 48 different peripheral channels. Each clock generator can be driven by one of 8 different internal clock sources or by an external clock supplied on one of 2-3 different IO pins. Each of the 48 peripheral channels can be configured to accept a clock from any of the 12 different generators.

It seems I ought to parameterize the module in terms of the three different pieces: clock sources, clock generators, and peripheral channels. One of my goals for the module is to make it impossible for users to accidentally change a clock that is being used by some generator or peripheral. I thought this would be possible using the Rust ownership system, but it's not as ergonomic as I would like.

Each clock source has some associated resource that controls or supplies the clock. For external clocks, the resource is the `gpio::Pin`. For internal clocks, this could either be a HAL `struct`, if it exists, or a PAC `Reg` or `RegisterBlock`. When creating a `clock::Source` struct, I want it to take ownership of the associated resource, so that the resource cannot be modified or dropped while the clock source is in use. That is simple enough.

A similar situation exists between clock sources and clock generators. I don't want to allow users to `free` a `clock::Source` while a `clock::Generator` is still using it. If the clock source is external, then the clock source can only be used for a single clock generator. In that case, the `clock::Generator` could own the `clock::Source`, which would be simple. However, the situation is different for internal clock sources. In that case, any number of generators could use the same `clock::Source`. My first instinct was to use `&Source`, or more accurately `PhantomData<&'a Source>`, to keep it zero-sized. But that creates a problem. Once you create the `clock::Source` and supply a reference to the `clock::Generator`, you can never move the `clock::Source`. To me, this is very constraining. For instance, if you were using `RTIC` and you set up the `clock::Source` in `init`, you would probably want to move it into `Resources` for use by some other task(s). But once the `clock::Generator` has been created, you can no longer move the `clock::Source`.

I think my problem boils down to the difference between zero-sized types and normal types. If the `clock::Source` had a size greater than zero, this problem would be unavoidable. You would *have* to store the `clock::Source` somewhere off the stack before referencing it. But in this case, each `clock::Source` is guaranteed to be zero-sized. As a result, there is no memory to point to, so there is no reason you couldn't move the `clock::Source` after referencing it. Really, I'm trying to use the reference as a way to guarantee the `clock::Source` outlives the `clock::Generator`, but nothing more. Can anyone think of a way to implement that?

I tried using a technique described in the [zero-sized references](https://ferrous-systems.com/blog/zero-sized-references/) article. The approach there is to use a `static mut` allocation within the `impl` of a zero-sized type to store the resource in question. But that doesn't work for me, because I need the allocation to be generic. Here's an outline of what I was trying to do, but it won't compile on the Playground.

```rust
use std::marker::PhantomData;
use std::mem::MaybeUninit;

pub struct Register;

pub trait SourceType: Sized {
    type Resource;
    fn new(resource: Self::Resource) -> Source<Self> {
        Source::new(resource)
    }
}

pub struct SourceTypeA;

impl SourceType for SourceTypeA {
    type Resource = Register;
}

// Now that each Source can be copied, you can move it into each generator
#[derive(Copy, Clone)]
pub struct Source<S>
where
    S: SourceType,
{
    marker: PhantomData<S::Resource>,
}

impl<S> Source<S>
where
    S: SourceType,
{
    fn new(resource: S::Resource) -> Source<S> {
        let s = Self { marker: PhantomData };
        *s.storage() = resource;
        s
    }
    // Note: I haven't thought through all of the issues with unsafe here.
    // This was just a quick test.
    fn storage(&self) -> *mut S::Resource {
        static mut STORAGE: MaybeUninit<S::Resource> = MaybeUninit::uninit();
        unsafe { STORAGE.as_mut_ptr() }
    }
    // This is not correct as written, because each Source copy could call free.
    // I haven't thought through how to solve that problem,
    // because I didn't get past the `static` problem.
    pub fn free(self) -> S::Resource {
        *self.storage()
    }
}
```

This example gives me `error[E0401]: can't use generic parameters from outer function`. I tried promoting the `static` allocation to the `impl` level, but you can't have `static` items in an `impl`. The same restriction applies to traits as well.

I suppose I could manually allocate all of the necessary `static`s at the module level and point to the correct `static` in each `SourceType` implementation, but that feels really hacky. Is there a better way?

blm768, 2020-10-01:
One option might be to create a separate struct (or a separate typestate) representing a configured, "frozen" clock that can be used by a peripheral. For clock sources that can be shared, the frozen clock handle would implement `Clone` (or an analogous operation), and peripherals could take ownership of the clones,, with all clones implicitly sharing ownership of the clock source (without actually holding a reference to it). The tricky part would be taking ownership of the clock back so it could be unfrozen. You could implement that a couple of ways:

- Store a reference count in a `static` `core::sync::atomic::AtomicU8`. Increment it every time you create a frozen handle, and decrement it with `Drop`. Have the frozen handle implement `TryInto<UnfrozenClockSource>`, where `TryInto` only succeeds if the reference count is one.

  This requires taking up some memory at runtime, but not a lot.
- Store a reference count as a type parameter, i.e. by using the `typenum` crate. "Cloning" a handle consumes it and produces two new handles with a higher reference count. Handles with reference counts greater than one can be "merged", reducing the count again. Handles with a reference count of one can be unfrozen.

  I'd have to think through the full implications of this strategy, since the reference count wouldn't be global. This strategy would also make it harder to reclaim ownership in cases where the structure of the code doesn't let the compiler "prove" that everything will be reclaimed.

blm768, 2020-10-01:
Actually, that latter idea probably won't work as-is, at least not without some annoying restrictions. Something like this rough sketch might work, though. `AddOne`, `SubtractOne`, and `Zero` don't actually exist as-is, but I'm too lazy to read the `typenum` docs right now to figure out what they would actually be.

```rust
struct Source {}

impl Source {
    pub fn freeze() -> Frozen;
}

struct Frozen<RefCount> {}

impl<RefCount> Frozen<RefCount> {
    pub fn borrow(self) -> (Self<AddOne<RefCount>>, Handle>);
    pub fn return(self, handle: Handle) -> Self<SubtractOne<RefCount>>, Handle>;
}

impl Into<Source> for Frozen<Zero> {
    // ...
}
```

A downside of this approach is that you can't just rely on `Drop`; any dropped handle is lost forever, so you can never use that clock source again.

bradleyharden, 2020-10-01:
Hmm. Good idea. I was thinking that I wanted an `Rc`, but I didn't want to pay the runtime cost for an `Rc`. So create a type-level `Rc`! Here's something that compiles on the Playground.

```rust
use std::marker::PhantomData;
use std::ops::{Add, Sub};
use typenum::{Unsigned, NonZero, Add1, Sub1, B1, U0}; // 1.12.0

struct TypeRc<N: Unsigned> {
    n: PhantomData<N>,
}

impl<N> TypeRc<N>
where
    N: Unsigned + Add<B1>,
    Add1<N>: Unsigned,
{
    pub fn borrow(self) -> TypeRc<Add1<N>> {
        TypeRc { n: PhantomData }
    }
}

impl<N> TypeRc<N>
where
    N: Unsigned + NonZero + Sub<B1>,
    Sub1<N>: Unsigned,
{
    pub fn replace(self) -> TypeRc<Sub1<N>> {
        TypeRc { n: PhantomData }
    }
}

impl TypeRc<U0> {
    pub fn free(self) {}
}

```

bradleyharden, 2020-10-02:
Actually, I guess your first version of that idea was close to an `Rc`, but like you said, it's a bit awkward, if it would even work. I think the latter approach is better. It's reference counting, but it's not an `Rc`.

bradleyharden, 2020-10-05:
I'm pretty much done with my refactor of the `clock` module. I don't have time to write it up tonight, but I will try to soon.

I implemented compile-time reference counting, as discussed above. However, the current implementation doesn't use any abstraction or encapsulation. It directly implements the code above, which is not the prettiest. Separately from the `clock` module, I've been trying to find some way to encapsulate compile-time reference counting, but I have struggled to find an implementation that significantly reduces the number of trait bounds needed in user code. I kept hitting road block after road block. Tonight, I finally came up with a clean solution. I thought I would leave it here. I'll probably update my `clock` code to use this encapsulation.

Here's the library code.

```rust
use std::marker::PhantomData;
use std::ops::{Add, Sub};

use typenum::{Unsigned, Bit, Add1, Sub1, UInt, B1, U1, U0};

struct RefCountType<N: Unsigned> {
    n: PhantomData<N>,
}

pub trait UNonZero: Unsigned {}

impl<U: Unsigned, B: Bit> UNonZero for UInt<U, B> {}

pub trait RefCount {}

impl<N: Unsigned> RefCount for RefCountType<N> {}

pub trait RefCountOps {
    type Add: RefCount;
    type Sub: RefCount;
    fn add(self) -> Self::Add;
    fn sub(self) -> Self::Sub;
}

impl<N> RefCountOps for RefCountType<N>
where
    N: UNonZero + Add<B1> + Sub<B1>,
    Add1<N>: UNonZero,
    Sub1<N>: Unsigned,
{
    type Add = RefCountType<Add1<N>>;
    type Sub = RefCountType<Sub1<N>>;
    fn add(self) -> Self::Add {
        RefCountType { n: PhantomData }
    }
    fn sub(self) -> Self::Sub {
        RefCountType { n: PhantomData }
    }
}

impl RefCountOps for RefCountType<U0> {
    type Add = RefCountType<U1>;
    type Sub = RefCountType<U0>;
    fn add(self) -> Self::Add {
        RefCountType { n: PhantomData }
    }
    fn sub(self) -> Self::Sub {
        RefCountType { n: PhantomData }
    }
}

pub trait RefCountZero: RefCount {}

impl RefCountZero for RefCountType<U0> {}
```

And here is user code.

```rust
pub struct Test<T>
where
    T: RefCount,
{
    t: T,
}

impl<T> Test<T>
where
    T: RefCount + RefCountOps,
{
    pub fn borrow(self) -> Test<T::Add>
    {
        Test { t: self.t.add() }
    }
    pub fn unborrow(self) -> Test<T::Sub>
    {
        Test { t: self.t.sub() }
    }
}

impl<T: RefCountZero> Test<T> {
    pub fn free() {}
}
```

I really like how it turned out. The user code is much cleaner, and users don't need to import or understand anything about `typenum`.

bradleyharden, 2020-10-05:
Hmm. I ran into another hang-up with the `clock` module. Maybe you can help @blm768.

I liked the idea of storing some object that controls the clock within each `clock::Source` struct. I think that makes a lot of sense for external clocks, because you can store and release the `gpio::Pin`. However, I just realized that structure doesn't work well for most, if not all, of the other clock sources.

At first, I wanted to store one of the PAC control registers within the `clock::Source`, but then I realized that wouldn't work. For instance, the `Peripherals.OSCCTRL` struct is zero-sized and can be moved out of `Peripherals`, but that struct controls multiple clocks. To get finer grained control, I thought about using individual registers, like `Peripherals.OSCCTRL.dfllctrla` for the DFLL source. But then I realized that the register itself is sized and exists behind `OSCCTRL` which implements `Deref` to point to the correct location in memory. You can't move `dfllctrla` out of `OSCCTRL`, and it wouldn't even make sense if you could, because `dfllctrla` is a `VolatileCell` underneath.

Furthermore, clock generators 2 - 11 can use generator 1 as a clock source. I wouldn't want to store generator 1 inside a `clock::Source`, because it can still be used for other purposes.

Ideally, I would like to prevent accidental changes to the clock sources at compile-time. But I'm not sure how to do that with my current architecture.

Instead, maybe I could create singleton tokens that represent each clock source. Then, when you go to configure each clock source elsewhere in the HAL, you would need to provide that token, indicating that the clock is not in use.

I think that might actually work well, but it requires HAL interfaces for each clock source to make it safe. I'm probably not willing to implement interfaces for every clock source type myself, since I would like to focus on the SPI and UART modules. But maybe others are interested and can implement them instead.

Thoughts?

blm768, 2020-10-09:
This sounds a lot like what I've been playing with in terms of s[plitting `SERCOM`s](https://github.com/atsamd-rs/atsamd/pull/240). Basically, the PAC-level object is moved into a container type (in this case, `Sercom`, which can then be split into multiple parts  either by calling an appropriate method or moving out of public `struct` fields. The PAC object is dropped at that point, but the individual split objects can use `unsafe` code to access the underlying registers (with the understanding that no two split objects will access the registers in a way that might step on another object's toes).

bradleyharden, 2021-06-14:
A lot of the ideas in this thread overlap with #429 and are included in #450. I'm going to close this, finally.

closed 2021-06-14 by bradleyharden