atsamd-rs/atsamd #287: atsamd-rs documentation site seems to be down

TDHolmes, 2020-10-24:
The documentation link on atsamd-rs on [crates.io](https://crates.io/crates/atsamd-hal) points to https://atsamd-rs.github.io, which seems to be down.

sajattack, 2020-10-24:
Yup, these links are old and should be removed for docs.rs links to take over.

TDHolmes, 2020-10-24:
[atsamd-hal docs.rs](https://docs.rs/crate/atsamd-hal/0.10.0) seems to not build successfully. Will that be fixed first?

sajattack, 2020-10-24:
The HAL documentation doesn't build without a specific target board.

jessebraham, 2020-10-25:
The documentation hosted on [docs.rs](https://docs.rs/atsamd-hal/0.10.1/atsamd_hal/) builds now. Since a chip must be specified, the `samd21g` is being used, as was the case with the old documentation.

closed 2020-10-25 by jessebraham